intel/media-driver #980: Please don't mention i915.force_probe in end-user documentation

danvet, 2020-06-17:
Ping @XinfengZhang, we discussed in private a bit.

Preliminary hw support is only for UMD developers, never for end-users. As soon as end-users discover this option, they will use it, and upstream linux kernel will force us to provide full regression fixes support even for code we know isn't ready. That's why we have to rename this option every few years already because too many users have discovered it again.

Therefore please remove it from end-user documentation (especially the main README file), and hide it somewhere where only UMD developers will discover it. Same goes for kernel support matrices, preliminary support means "no support" for end-users.

XinfengZhang, 2020-07-01:
close it, already was fixed by bcfaf0b43e02d14b373995523c1475d07442c1f9

closed 2020-07-01 by XinfengZhang

danvet, 2020-07-01:
Thanks for addressing this so quickly!