intel/media-driver #985: A media driver patch causes gst-vaapi pass rate dropped on TGL

xinyudox, 2020-06-23:
this patch:

commit 50c845460f3c2453afb0f965618bc6940a434e6c
Author: Bai, Guangyao <guangyao.bai@intel.com>
Date:   Wed Jun 10 16:40:13 2020 +0800

    [Media Common] Defaule Enable Media MMC
    
Defaule Enable Media MMC for BW saving

egg:

command line:
gst-launch-1.0 -vf filesrc location=SA00040.vc1 ! 'video/x-wmv,profile=(string)advanced',width=176,height=144,framerate=14/1 ! vaapivc1dec ! videoconvert dither=0 ! video/x-raw,format=I420 ! checksumsink2 file-checksum=false qos=false frame-checksum=false plane-checksum=false dump-output=true dump-location=./test.yuv

log:
DETAIL: ssim = [0.2966, 0.0791, 0.0807, 0.5542, 0.3995, 0.3947]
AssertionError: assert 0.2966 >= 0.99

XinfengZhang, 2020-07-01:
this patch enabled MMC, so, there are some impact with memory sharing cases. but should not impact legacy feature. @Guangyao-Bai , please help to take a look

Guangyao-Bai, 2020-07-01:
yes, already have a fix, check-in WIP

XinfengZhang, 2020-08-07:
there are a serious patch to fix the issue, last one is 84a1423a74c7ee82a5b397f61c6eac66c717b754.

closed 2020-08-07 by XinfengZhang