intel/media-driver #988: Regression: Display corruption for NV12 format with dmabuf

VUNNYSODHI, 2020-06-29:
**Command Used:**
$GST_GL_PLATFORM=egl gst-launch-1.0 videotestsrc ! video/x-raw,format=NV12,width=1920,height=1080 ! msdkvpp ! 'video/x-raw(memory:DMABuf),format=NV12' ! glimagesink

**Platform:** TGL

**OS**:  Ubuntu 18.04.3 LTS

**Environment:** 
Latest Libva: https://github.com/intel/libva/commit/3cc2212c38630ffcdc6b38e0bd867845adee5ed9
Latest Gmmlib: https://github.com/intel/gmmlib/commit/40c7d14fd16a4282410a2f6b0cd59251c8e41d5c 
Latest MediaDriver: https://github.com/intel/media-driver/commit/adc23261d4a4f60d6e113959cdb3acbed055fde9 

**Output**:
[image]

VUNNYSODHI, 2020-06-29:
Issue can be resolved if we revert 2 patches https://github.com/intel/media-driver/commit/cc869f7a27c0fbf7182ecca8e8a3e2fb383e233a ([VP] Fix KW issue)+ https://github.com/intel/media-driver/commit/a88a4d60fa8f67531d121c43b28f883b12ed3af0 ([Media Common] Fix Surfacing sharing issue on Linux) on top of adc2326

XinfengZhang, 2020-07-01:
the issue should be related with MMC,  need to check it about modifier support, @Guangyao-Bai , could you take a look?

Guangyao-Bai, 2020-07-01:
Media Alreay support Modifier and we passed with Kodi Player. what APP are you using? if Gstreamer or FFMPEG, then it need to switch to Memory_type_2 for surface info sharing

VUNNYSODHI, 2020-07-02:
Yes @Guangyao-Bai i am using gstreamer, can you please explain more on switching to Memory_type_2?

xhaihao, 2020-07-03:
@VUNNYSODHI please do not use DMABuf on TGL or revert MMC in your environment.  Currently DMABuf in gstreamer doesn't support modifier, however to support MMC, modifier is required.

VUNNYSODHI, 2020-07-03:
Thanks for the info @xhaihao, currently anyone working on adding MMC modifier support in gstreamer?

xhaihao, 2020-07-03:
@VUNNYSODHI It seems MMC is disabled in the latest media driver. Your pipeline works well for me on my TGL with the latest media-driver (commit 0c96976ab972f7633ea205aedd4177a0a42b4fab)

So far, someone did some initial work for modifier in waylandsink https://gitlab.freedesktop.org/gstreamer/gst-plugins-bad/-/merge_requests/470. we need similar work in glimagesink.

I filed two issues (https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/issues/263 and https://gitlab.freedesktop.org/gstreamer/gst-plugins-bad/-/issues/1329) for modifier support in gstreamer-vaapi and gst-msdk.

Guangyao-Bai, 2020-07-03:
@xhaihao @VUNNYSODHI  yes, MMC was disable for non-production board. but when you switch to production, it will enabled automatically.

VUNNYSODHI, 2020-07-06:
Thanks @xhaihao and @Guangyao-Bai for info.
I have retested with https://github.com/intel/media-driver/commit/0c96976ab972f7633ea205aedd4177a0a42b4fab + latest libva(afca400), gmmlib(40c7d14),  on my setup

**Still see corruption on my side**

**Addition info:**
libva info: VA-API version 1.8.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_8
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.8 (libva 2.8.0.1)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 20.3.pre (0c96976a)

XinfengZhang, 2020-12-31:
the policy changed , MMC is disabled by default now , so , suppose the issue already been fixed by latest media-driver

VUNNYSODHI, 2021-01-04:
Thanks @XinfengZhang, not able to reproduce issue with latest media-driver tip. Hence closing ticket.

closed 2021-01-04 by VUNNYSODHI