intel/media-driver #989: not supported multi transcode

xinyudox, 2020-07-01:
commond line:
gst-launch-1.0 filesrc location= /media/H264_Videos/HoneyBee_1920x1080_10mbps_30fps_Baseline_at_L4.2.mp4 ! qtdemux ! h264parse ! vaapih264dec ! tee name=t ! vaapih264enc tune=low-power ! h264parse ! qtmux ! filesink location= /tmp/HoneyBee_1920x1080_10mbps_30fps_Baseline_at_L4.2_multi-transcoded_h264.mp4 t. ! vaapih265enc tune=low-power low-delay-b=true ! h265parse ! matroskamux ! filesink location=/tmp/HoneyBee_1920x1080_10mbps_30fps_Baseline_at_L4.2_multi-transcoded_h265.mkv t. ! vaapijpegenc ! avimux! filesink location=/tmp/HoneyBee_1920x1080_10mbps_30fps_Baseline_at_L4.2_multi-transcoded_jpeg.avi

Running this command will print "Encoding failure due to HW issue"

XinfengZhang, 2020-07-01:
is there details, such as HW platform, vainfo messages, logs?

xuguangxin, 2020-07-08:
@XinfengZhang any update on this?
thanks

Sherry-Lin, 2020-07-14:
Could you please confirm whether it's workable for signal transcode in EHL with 5.4 kernel? As I know, EHL also require 5.7+ kernel version.

xinyudox, 2020-07-14:
it's workable for signal transcode in EHL with 5.4 kernel.

xuguangxin, 2020-07-17:
@Sherry-Lin any update on this?
thanks

Xiaogangli-intel, 2020-08-12:
@xinyudox we checked this issue, the failure is caused by JPEG encoding, although with single JPEG encoding, it still failed.

But you mentioned, "it's workable for signal transcode in EHL with 5.4 kernel.", single H264? H265 or JPEG?

XinfengZhang, 2020-08-13:
suppose it is same issue with #902 ?

zcwang, 2020-08-17:
@XinfengZhang could you please to check issue #1030 for similar issue comparing to this issue? Thanks!

Xiaogangli-intel, 2020-08-19:
@xinyudox Could you upgrade your Linux kernel?
The requirement already mentioned in README. (EHL/JSL/TGL: drm-tip, missed in <=5.4.3 kernels)

JPEG encoding works well if we don't run AVC encoding. So maybe some of the registers were not reset after run AVC encoding. This maybe related to KMD, so please try the KMD which supports EHL.

zcwang, 2020-09-17:
@xinyudox May I know if it still could be reproduced after upgrade kernel to newer version >= 5.4.3 for my reference?

Thanks!
Gary

xinyudox, 2020-09-18:
This issue can't be reproduced using kernel version 5.9.0-rc2+.

closed 2020-09-18 by xinyudox