intel/media-driver #990: VAConfigAttribPredictionDirection query failed on VMEPAK mode.

HeJunyan, 2020-07-01:
The VAConfigAttribPredictionDirection query for HEVC encoder in VMEPAK mode on TGL only returns VA_ATTRIB_NOT_SUPPORTED, which causes that the low-delay-b mode can not be detected correctly. 
"VA_PREDICTION_DIRECTION_PREVIOUS | VA_PREDICTION_DIRECTION_FUTURE | VA_PREDICTION_DIRECTION_BI_NOT_EMPTY" should be the correct result.
For VDENC mode, it works well.

And in the code, I notice that this property is only set in VAEntrypointEncSliceLP of HEVC, and is not supported for all VMEPAK mode and most of the other encoders.

Please refine this property to make sure the query result is correct for all platforms, all profiles and all entrypoints. At least, for the gen11+ platforms which has low-delay-b feature.

XinfengZhang, 2020-07-01:
thanks reporting, will file PR to fix it soon

HeJunyan, 2020-08-06:
I checked the newest code in master, and find #1009 does not fix all problems on TGL. Though, the ICL can works well.

```
VAStatus MediaLibvaCapsG12::QuerySurfaceAttributes(
        VAConfigID configId,
        VASurfaceAttrib *attribList,
        uint32_t *numAttribs)
{
   .......

    if(entrypoint == VAEntrypointEncSliceLP)
    {
     .....

        else if (IsHevcProfile(profile))
        {
            attrib.type = (VAConfigAttribType) VAConfigAttribPredictionDirection;
            attrib.value = VA_PREDICTION_DIRECTION_PREVIOUS | VA_PREDICTION_DIRECTION_FUTURE | VA_PREDICTION_DIRECTION_BI_NOT_EMPTY;
            (*attribList)[attrib.type] = attrib.value;
        }
    }

Only in VDENC mode, this attribute is set, but in VMEPAK mode, this attribute is still unset.

And so far as I know, on TGL, the VMEPAK mode  also does not support P frame, so this still cause our pipeline hang:

gst-launch-1.0 -v -f videotestsrc num-buffers=100 ! capsfilter caps=video/x-raw,format=NV12,width=400,height=320 ! vaapih265enc ! video/x-h265,profile=main ! h265parse ! filesink location=out.265

HeJunyan, 2020-08-06:
I can not find the reopen option, @XinfengZhang please reopen it.

XinfengZhang, 2020-08-07:
@HeJunyan , please double confirm it, the code you copied should be old , VME & VDEnc is same on TGL, support I B Lowdelay B

XinfengZhang, 2020-08-07:
@HeJunyan , could you help to check whether #1021 fix the issue?

HeJunyan, 2020-08-07:
It works for me on my TGL, please help to merge it.

commit e5c78354 (2020-08-04): correct the reference capibility for gen11 and gen12

Fixes #880,  Fixes #990
PR #1021 (2020-08-11): correct HEVC frame type capability of TGL HEVC support I B lowdelay B
Fixes #990
commit a51bd7c3 (2020-08-11): correct HEVC frame type capability of TGL HEVC support I B lowdelay B

Fixes #990
closed 2020-08-11 by intel-mediadev