intel/media-driver #997: TGL Surface sharing methodology update on TGL

Guangyao-Bai, 2020-07-09:
TGL will use memory type VA_SURFACE_ATTRIB_MEM_TYPE_DRM_PRIME_2 for surface sharing with compression/tiling etc. info. thus may need outside user upgrate the memory type supporting logic when usages include more than 2 component like media + Mesa.

Sherry-Lin, 2020-07-09:
This interface upgrade is necessary in DMA buffer sharing for ffmpeg/gstreamer. Detail information as below.
* https://github.com/intel/libva/issues/424
* [gst-msdk/issues/1329](https://gitlab.freedesktop.org/gstreamer/gst-plugins-bad/-/issues/1329)
* [gstreamer-vaapi/issues/263](https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/issues/263)
* [ffmpeg-vaapi patch](https://patchwork.ffmpeg.org/project/ffmpeg/patch/20200619064733.146001-2-haihao.xiang@intel.com/)

Sherry-Lin, 2020-07-09:
If you want to know more about compression info sharing, you could access https://github.com/intel/media-driver/wiki/Compression-Info-Sharing

Sherry-Lin, 2020-07-09:
If you have to disable memory compression, you could revert https://github.com/intel/media-driver/commit/50c845460f3c2453afb0f965618bc6940a434e6c \(https://github.com/intel/media-driver/pull/970 \), but the driver health is not qualified in test.

closed 2020-10-25 by XinfengZhang