intel/media-driver #998: vaQueryImageFormats does not return Y212 and Y412 format on TGL.

HeJunyan, 2020-07-09:
Y212 and Y412 formats should be return if we want to decode 12 bits 422 and 444 streams in HEVC.
The Y216 and Y416 are returned, but we need to use 12 bits format, it is more precise. And I also notice that the P012 is available. 

Please also make sure Y212 and Y412 are in the list of vaQuerySurfaceAttributes's VASurfaceAttribPixelFormat, that can make these formats as RT surface format of encoder/decoder context.

wangyan-intel, 2020-07-14:
@HeJunyan It should be similar with #921. Right? I can add them.

HeJunyan, 2020-07-14:
Yes, please. And I notice the P012 is ready.

wangyan42164, 2020-07-14:
@HeJunyan Could you please try #1002 which is dependent on https://github.com/intel/libva/pull/434? Thanks.

FurongZhang, 2020-07-20:
Most of code changes are related to Codec.

PR #1002 (2020-09-04): Add Y212 and Y412 format
Fixes #998.
commit be8a2cf7 (2020-09-04): Add Y212 and Y412 format for decoding

Fixes #998.
closed 2020-09-04 by intel-mediadev