intel/media-driver #999: [FFmpeg][HEVC][SCC][Decode]

fulinjie, 2020-07-09:
FFmpeg:
https://github.com/intel-media-ci/ffmpeg/pull/231

Note that Screen-Extended Main 4:4:4 and 4:4:4 10 supports chroma_format_idc from 0, 1 or 3, hence both 420 and 444 are supported.

However, in MediaLibvaCapsG12::QuerySurfaceAttributes (gen12) for SCCMain444, no 420 format is supported:

        else if(profile == VAProfileHEVCSccMain444)
        {
            //422/444 8/10bit included
            attribs[i].type = VASurfaceAttribPixelFormat;
            attribs[i].value.type = VAGenericValueTypeInteger;
            attribs[i].flags = VA_SURFACE_ATTRIB_GETTABLE | VA_SURFACE_ATTRIB_SETTABLE;
            attribs[i].value.value.i = VA_FOURCC_YUY2;
            i++;

            attribs[i].type = VASurfaceAttribPixelFormat;
            attribs[i].value.type = VAGenericValueTypeInteger;
            attribs[i].flags = VA_SURFACE_ATTRIB_GETTABLE | VA_SURFACE_ATTRIB_SETTABLE;
            attribs[i].value.value.i = VA_FOURCC_Y210;
            i++;

            attribs[i].type = VASurfaceAttribPixelFormat;
            attribs[i].value.type = VAGenericValueTypeInteger;
            attribs[i].flags = VA_SURFACE_ATTRIB_GETTABLE | VA_SURFACE_ATTRIB_SETTABLE;
            attribs[i].value.value.i = VA_FOURCC_AYUV;
            i++;

            attribs[i].type = VASurfaceAttribPixelFormat;
            attribs[i].value.type = VAGenericValueTypeInteger;
            attribs[i].flags = VA_SURFACE_ATTRIB_GETTABLE | VA_SURFACE_ATTRIB_SETTABLE;
            attribs[i].value.value.i = VA_FOURCC_Y410;
            i++;
        }

https://github.com/intel/media-driver/blob/master/media_driver/linux/gen12/ddi/media_libva_caps_g12.cpp#L1124

Summary:
Hence driver needs to add caps support for NV12 to enable SCC decoding support.

Here is a work around in FFmpeg for verifying:
https://github.com/intel-media-ci/ffmpeg/pull/231/commits/e442c218d3c30208ca9abce282db401b725fcbde

fulinjie, 2020-07-09:
Hi @XinfengZhang @Sherry-Lin, would you please help to check?

wangyan42164, 2020-07-14:
@fulinjie Could you please try #1001? Thanks.

PR #1001 (2020-07-15): Add NV12 format for HEVCSccMain444 based on the standard.
Fixes #999.
commit 4ab5939a (2020-07-15): Add NV12 format for HEVCSccMain444 based on the standard.

Fixes #999.
closed 2020-07-15 by intel-mediadev