intel/media-driver #1016: Regression: CPU side format conversions don't work anymore

eero-t, 2020-08-04:
Setup:
* HW: GEN9 (SkullCanyon)
* OS: Ubuntu 20.04
* Driver stack: latest Git version builds of drm-tip kernel, libdrm, gmmlib, libva, media-driver and FFmpeg

media-driver regressed between following commits:
*  2020-07-30 08:21:31 a732aa18a282770d3b3b13e58f54f568b721f5fe: [VP] Refine VP HW Filter
* 2020-08-02 11:43:13 8aa1d74b80668f9963e691b1c01ab564f50aec85: media: don't advertise AV1 decoding if the platform doesn't support it

CPU side format conversions don't anymore work, for example:
`ffmpeg -y -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i <AVC-file>.264  -vf hwdownload,format=uyvy422 -f null -`

Gives now following error, since the latter commit:
```
[AVHWFramesContext @ 0x7f2f7c0581c0] Failed to read image from surface 0x1c: 2 (resource allocation failed).
[hwdownload @ 0x563a7e003d00] Failed to download frame: -5.
Error while filtering: Input/output error
Failed to inject frame into filter network: Input/output error
Error while processing the decoded data for stream #0:0
Conversion failed!
```
Whereas it works fine with the former commit (and AFAIK has worked fine since bug #757 was fixed).

Same issue happens for all CPU side format conversions, marked with "O" in the table here: https://trac.ffmpeg.org/ticket/7691#comment:9

XinfengZhang, 2020-08-13:
yes,  I reproduced the issue, but it already been fixed with latest driver. I tried a51bd7c36937c69c0cd824ec57c337dc27f6045a.

eero-t, 2020-08-13:
Thanks, verified with a51bd7c36937c69c0cd824ec57c337dc27f6045a.

closed 2020-08-13 by eero-t