intel/media-driver #1017: VP9 encoding on CML

AndrejOrsula, 2020-08-05:
Hello everyone, I would like a suggestion about what driver should I use to enable hardware-accelerated VP9 encoding on Comet Lake H-series. Thank you in advance.

## Setup

* **iGPU**: UHD Graphics (CML GT2) [i7-10875H]
* **OS**: Ubuntu 18.04
* **Kernel**: 5.4.0-42-generic
* **media-driver**: [intel-media-20.2.0](https://github.com/intel/media-driver/releases/tag/intel-media-20.2.0)
* **libva**: [2.8.0](https://github.com/intel/libva/releases/tag/2.8.0)
* **libva-utils**: [2.8.0](https://github.com/intel/libva-utils/releases/tag/2.8.0) (if at all relevant)
* **gmmlib**: [intel-gmmlib-20.2.3](https://github.com/intel/gmmlib/releases/tag/intel-gmmlib-20.2.3)

**Hardware support**: Looking at [Encode and Decode Capabilities for 7th Generation Intel® Core™ Processors and Newer](https://software.intel.com/content/www/us/en/develop/articles/encode-and-decode-capabilities-for-7th-generation-intel-core-processors-and-newer.html), the iGPU supports VP9 encoding.

## Issue

According to https://github.com/intel/intel-vaapi-driver/issues/499, Comet Lake is supported by this driver and I can no longer use `intel-vaapi-driver` anymore. I have already tested `media-driver` with both H.264 and H.265 in GStreamer 1.16.2, where everything seems fine. However, I am unable to use VP9 encoder as it is not a supported profile/entrypoint.

Below is the full output of `vainfo` and a partial `vainfo -a` for the available VP9 decoder profile/entrypoint pair.
```
$ vainfo
libva info: VA-API version 1.9.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_8
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.9 (libva 2.8.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 20.2.0 (97e2ea99)
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
```

```
$ vainfo -a
VAProfileVP9Profile0/VAEntrypointVLD
    VAConfigAttribRTFormat                 : VA_RT_FORMAT_YUV420
                                             VA_RT_FORMAT_YUV422
                                             VA_RT_FORMAT_YUV420_10
                                             VA_RT_FORMAT_RGB32
                                             VA_RT_FORMAT_YUV420_10BPP
    VAConfigAttribDecSliceMode             : VA_DEC_SLICE_MODE_NORMAL
                                             VA_DEC_SLICE_MODE_BASE
    VAConfigAttribDecProcessing            : VA_DEC_PROCESSING_NONE
    VAConfigAttribMaxPictureWidth          : 8192
    VAConfigAttribMaxPictureHeight         : 8192
    VAConfigAttribProcessingRate           : VA_PROCESSING_RATE_DECODE

VAProfileVP9Profile2/VAEntrypointVLD
    VAConfigAttribRTFormat                 : VA_RT_FORMAT_YUV420
                                             VA_RT_FORMAT_YUV422
                                             VA_RT_FORMAT_YUV420_10
                                             VA_RT_FORMAT_RGB32
                                             VA_RT_FORMAT_YUV420_10BPP
    VAConfigAttribDecSliceMode             : VA_DEC_SLICE_MODE_NORMAL
                                             VA_DEC_SLICE_MODE_BASE
    VAConfigAttribDecProcessing            : VA_DEC_PROCESSING_NONE
    VAConfigAttribMaxPictureWidth          : 8192
    VAConfigAttribMaxPictureHeight         : 8192
    VAConfigAttribProcessingRate           : VA_PROCESSING_RATE_DECODE

```

This output corresponds to the [Decoding/Encoding Features](https://github.com/intel/media-driver#decodingencoding-features) support matrix in the README of this repository, which states that only the decoder for VP9 is available for KBLx (under which CML is categorised).

So, what driver should I use to enable hw-accelerated VP9 encoding if `intel-vaapi-driver` does not support CML and only the VP9 decoder is available in `media-driver` for this platform? Any suggestions are welcome, thank you.

sreerenjb, 2020-08-05:
Not officially supported, but you can try this :https://github.com/intel/media-driver/pull/717

AndrejOrsula, 2020-08-06:
I can confirm that #717 works well on CML and I have not encountered any problems so far (tested with GStreamer). Thank you so much @sreerenjb for resolving my issue.

For more info, here is the new VP9 encoder profile/entrypoint pair.
```
$ vainfo -a
VAProfileVP9Profile0/VAEntrypointEncSlice
    VAConfigAttribRTFormat                 : VA_RT_FORMAT_YUV420
    VAConfigAttribRateControl              : VA_RC_CBR
                                             VA_RC_VBR
                                             VA_RC_CQP
                                             VA_RC_MB
    VAConfigAttribEncInterlaced            : VA_ENC_INTERLACED_NONE
    VAConfigAttribEncMaxRefFrames          : l0=8
                                             l1=0
    VAConfigAttribEncSliceStructure        : VA_ENC_SLICE_STRUCTURE_ARBITRARY_MACROBLOCKS
    VAConfigAttribMaxPictureWidth          : 4096
    VAConfigAttribMaxPictureHeight         : 4096
    VAConfigAttribEncJPEG                  : rithmatic_coding_mode=0
                                             progressive_dct_mode=0
                                             non_interleaved_mode=0
                                             differential_mode=0
                                             differential_mode=0
                                             max_num_components=4
                                             max_num_scans=1
                                             max_num_huffman_tables=2
                                             max_num_quantization_tables=3
    VAConfigAttribEncQualityRange          : number of supported quality levels is 7
    VAConfigAttribEncQuantization          : VA_ENC_QUANTIZATION_NONE
    VAConfigAttribEncIntraRefresh          : VA_ENC_INTRA_REFRESH_NONE
    VAConfigAttribEncSkipFrame             : supported
    VAConfigAttribProcessingRate           : VA_PROCESSING_RATE_ENCODE
    VAConfigAttribEncDirtyRect             : number of supported regions is 4
    VAConfigAttribEncParallelRateControl   : number of supported layers is 1
```

closed 2020-08-06 by AndrejOrsula

sreerenjb, 2020-09-22:
@ajgnet that patch was only for gen9.
You should have the vp9 encode support in gen10 with official iHD branch.

ajgnet, 2020-09-22:
@sreerenjb it's not. i wonder if the i7-10710U is simply not capable of VP9 encoding, for some reason.

sreerenjb, 2020-09-22:
Aha, okay, It seems to be the CML device. VP9 encode is not supported in the official tree then. Sorry for the confusion. In theory, it should work with my dev branch though. The gen9 gfx hardware has the dual pipe(using both EUs and FixedFunction unit) encode support.