intel/media-driver #1024: VP9 decoder for Skylake?

pedrocr, 2020-08-10:
According to this table:

https://en.wikipedia.org/wiki/Intel_Graphics_Technology#Hardware-accelerated_algorithms

as well as the Skylake source linked there:

http://www.intel.com/content/dam/www/public/us/en/documents/datasheets/desktop-6th-gen-core-family-datasheet-vol-1.pdf

Skylake should be capable of hardware decoding VP9. The old hybrid driver seemed to include some support for that. Is support in media-driver planned at all? It would be a boon for a bunch of older laptops trying to play youtube videos.

XinfengZhang, 2020-08-11:
SKL has no HW VP9 decoding support,  I suppose  you are talking about older intel-vaapi-driver driver, the solution is hybrid solution , if you want to use VP9 decoding support on SKL, I suggest still use https://github.com/intel/intel-hybrid-driver. media driver has no plan to add the support on SKL , media driver enable HW VP9 decoding from BXT/APL KBL

pedrocr, 2020-08-11:
I do mean that hybrid driver that now seems discontinued. Why not integrate that into this one?

XinfengZhang, 2020-10-25:
because the benefit is not enough :) hybrid solution always will consume CPU & GPU. no power gain, and perf gain is not enough

closed 2020-10-25 by XinfengZhang

runderwo, 2025-09-05:
The compatibility matrix in this project lists VP9 decode as available on SKL.  Is that incorrect?