intel/media-driver #1029: [TGL] Disable E2E Compression for hardware revisions from 0 to 2

ykku16github, 2020-08-17:
E2E compression is currently disabled for hardware revision 0.  See [1].  The kernel commit in [2] indicates that media decompression isn't available for steppings A0-C0.

If we test today's iHD with today's drm-tip on a B-stepping, for example, we would see corrupted video frames on the display by default.

Does it make sense to disable E2E compression until media decompression is available?

[1] https://github.com/intel/media-driver/blob/d457d66da7065e5116f818e4bca2d6ea4b841422/media_driver/linux/gen12/ddi/media_sku_wa_g12.cpp#L257
[2] https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=dbff5a8db9c630f61a892ab41a283445e01270f5

XinfengZhang, 2020-08-17:
hi @Guangyao-Bai , please help to check it

ykku16github, 2020-08-17:
@XinfengZhang , thanks for your quick response.  I submitted the following pull request:
https://github.com/intel/media-driver/pull/1031

Assuming the pull request makes sense, I have the following 3 additional questions:

1.  Should a similar change be made to the following?
https://github.com/intel/media-driver/blob/d457d66da7065e5116f818e4bca2d6ea4b841422/media_driver/linux/gen12/ddi/media_sku_wa_g12.cpp#L315

2.  Should a similar change be made to the following?
https://github.com/intel/media-driver/blob/d457d66da7065e5116f818e4bca2d6ea4b841422/media_driver/linux/gen12/ddi/media_sysinfo_g12.cpp#L139

3.  This pull request probably breaks any test case you have that assumes E2E compression is enabled by default.  How can that be addressed?

Guangyao-Bai, 2020-08-17:
1. The kernel commit in [2] in sharing about the stepping is for Display stepping, not for GT Stepping, from TGL, the GT/Display may have different Stepping info. for Example, the HW issue was fix on Display D Stepping while the GT stepping will map C Step.
2. based on talking with Display team, kernel can reject submision with such change if modifior with compression enabled info. but need to further check how they roll back ways. will check with Arches and get back.

ykku16github, 2020-08-17:
OK, thanks for the clarification.

XinfengZhang, 2021-01-08:
now , MMC already disabled by default on TGLB, so , I will close it. thanks

closed 2021-01-08 by XinfengZhang