intel/media-driver #1035: [csc][ffmpeg-vaapi]Failed to convert ayuv to another format

zhangyuankun-star, 2020-08-21:
**cmd-line**
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -pix_fmt 0yuv -s:v 1280x720 -i /home/yefeng/build/media/yuv/zyk_csc_1280x720_AYUV.yuv -vf 'hwupload,scale_vaapi=format=yuv420p,hwdownload,format=yuv420p' -f rawvideo -vsync passthrough -an -vframes 10 -y 1.yuv
**error**
[Parsed_scale_vaapi_1 @ 0x55a396551c00] Failed to render parameter buffer: 18 (invalid parameter).
Error while filtering: Input/output error
Failed to inject frame into filter network: Input/output error
Error while processing the decoded data for stream #0:0

I found that before the execution of the command fails, the function IS_YUV_FORMAT(format) will judge whether ayuv is yuv or not, but there is no ayuv format in the list that is judged to be true. I try to fix this issue by https://github.com/intel/media-driver/pull/994 , but it will cause regression. 
I hope you can help me fix it , thanks~

FurongZhang, 2020-08-25:
Thanks @zhangyuankun-star . Sure.

FurongZhang, 2020-08-25:
There is one PR https://github.com/intel/media-driver/pull/994 to fix the above issue. But it causes regression. Need some one to take a look again.

XinfengZhang, 2020-10-21:
#994 was merged, @zhangyuankun-star  could you help to check whether it is fixed?

MicroYY, 2020-12-15:
This issue is fixed by https://github.com/intel-innersource/drivers.gpu.unified/pull/4091
Could you pls verify it? @zhangyuankun-star

zhangyuankun-star, 2020-12-15:
The failed cases in CI can be passed successfully now, so this issue has been solved. Thank you very much!

closed 2020-12-15 by zhangyuankun-star