intel/media-driver #1037: vaDeriveImage()/vaGetImage() doesn't fill the last column and row in chroma plane if the dimension is odd

rosetta-jpn, 2020-08-28:
This is same issue of i965 driver filled in https://github.com/intel/intel-vaapi-driver/issues/516.
vaDeriveImage() in media driver has the same issue unlike i965 driver though.
I expect the same gstreamer command can reproduce this issue.

https://github.com/intel/intel-vaapi-driver/issues/516#issuecomment-672166170

sreerenjb, 2020-08-28:
Issue can be reproduced with slightly different command:
 gst-launch-1.0 videotestsrc num-buffers=90 !  video/x-raw, width=320, height=240 ! vaapipostproc width=321 height=241 ! xvimagesink

sreerenjb, 2020-08-28:
@XinfengZhang  Can you please find someone to look into this issue?

XinfengZhang, 2020-08-31:
@FurongZhang , there are VPP call from 320x240 to 321x241. could you help to to check it?
@sreerenjb, @rosetta-jpn  the command line shows it is a VPP issue, but not sure whether we are also discussing odd resolution decoding or encoding? we already have a similiar fix for VP9 encoding  https://github.com/intel/media-driver/commit/d6166b3538a89376771f2769d49d1ade8b177f7e.

rosetta-jpn, 2020-09-01:
I think decoding should be fine. I remember the video on screen looked fine when I played odd dimension video with iHD driver.

FurongZhang, 2020-09-01:
May I know why you need to do 320x240 to 321x241? It is only one pixel difference.

sreerenjb, 2020-09-01:
@FurongZhang one pixel difference is **big** :)
It is not just about 320x240 to 321x241, there are plenty of odd dimension videos out there and because of wrongly implemented vaGetImage (==vpp internally) in the driver, we see green lines in the output.

rosetta-jpn, 2020-09-15:
Any update on this bug?

FurongZhang, 2020-09-15:
Basically, we expect surface should be at least 2 aligned. What is your format? As I know, at least for NV12, we can not support odd resolution. What is the justification and impact from your side?

XinfengZhang, 2020-09-16:
@FurongZhang , I suggest to rounding up the alignment inside driver  ,such as 321x241 to 322 x 242, of course, application still treat it as 321x241 . suppose it will fix the issue .  @rosetta-jpn , does it resolve the issue?

sreerenjb, 2020-09-16:
@FurongZhang We can't change the alignment for each driver, in fact, this is working fine with i965 driver now. Also it is not possible to go and change all the middleware stacks like gstreamer, ffmpeg etc unless it is a VA spec change.

Jister, 2020-09-24:
As I tried the following command: gst-launch-1.0 videotestsrc num-buffers=90 ! video/x-raw, width=320, height=240 ! vaapipostproc width=321 height=241 ! filesink location=./test.yuv 
For 320x240 to 321x241, VPP will scale the image to 322x242 by default, and the last colum(321) and row(241) will be filled with data. But as I check the filesink output, the app save the image to 324x242, is it correct? 
[test_0924.zip](https://github.com/intel/media-driver/files/5274162/test_0924.zip)

rosetta-jpn, 2020-10-06:
Any update on this issue?

sreerenjb, 2020-10-06:
@Jister saving the raw data with filesink might not work in all cases, because it won't respect the cropped rectangle or additional padding bytes set in the memory. 
I usually use this one https://github.com/sreerenjb/gst-checksumsink for dumping the raw data to file.

XinfengZhang, 2020-10-10:
@sreerenjb , just like @Jister  said, VPP will align the surface to 322x242, and all data will be filled. so, there should not be any green line , if there are greeline, seems the display resolution > 322x242. filesink shows it is 324x242. xvimagesink maybe also is > 322x242.  you mean gst-checksumsink will show it to be <=322x242?

sreerenjb, 2020-10-12:
That is still not answering the question why display getting that green line only with ihd but not with i965 :)
BTW, the filesink pipeline & xvimagesink pipeline are using different formats.
The filesink pipeline will save the raw data as nv12 which is of course the intel's native format. On the other hand, xvimagesink (or try with other software renderer like ximagesink with proper csc) will use the i420 because it can't render nv12. So vpp is converting from nv12 to i420 in this pipeline.

FurongZhang, 2020-10-13:
Hi, Regarding the issue https://github.com/intel/media-driver/issues/1037, please see the dump YUV file. VPP has correct behavior. VPP output the correct value in X=321. 

For this case, App passed a output surface with 4 alignment, VPP processing with 2 alignment(But VPP processing all the valid region). When displaying, App needs to specify the valid display region. How about setting  output_region  in App level (https://github.com/intel/libva/blob/master/va/va_vpp.h)?  Please respond ASAP, I am trying to close this issue ASAP. 

From the below, we can see X=321, Y = 78, The pixel value RGB is (0, 0, 207). It is not the green line.   

The dumped file resolution is 324x242. But the X,Y 323, 32, the RGB value is 0,135, 0. It is green line.  

Thanks,
Furong

FurongZhang, 2020-10-13:
[image]

FurongZhang, 2020-10-13:
[image]

FurongZhang, 2020-10-13:
[image]

XinfengZhang, 2020-10-13:
> That is still not answering the question why display getting that green line only with ihd but not with i965 :)
I have not test i965, maybe i965 already aligned to 324x 244? so there are no green line?
> BTW, the filesink pipeline & xvimagesink pipeline are using different formats.
> The filesink pipeline will save the raw data as nv12 which is of course the intel's native format. On the other hand, [...]
I believe it is same with both nv12 and i420, VPP should already  aligned to 322x242 and fill the values. but beyond this rectangle  , it should be green. could you help to verify the display region?

sreerenjb, 2020-10-13:
Hm, why am I getting the green line even with ffmpeg too? Both ffmpeg & gst are asking to display only 321x241 , at least in my analysis.

gst-launch-1.0 filesrc location= sample_vp8.webm ! matroskademux ! vaapivp8dec ! vaapipostproc width=321 height=241 format=i420 ! checksumsink2 dump-output=true dump-location=sample_check_i420.yuv

ffplay -f rawvideo -pixel_format yuv420p -video_size 321x241 sample_check_i420.yuv

sreerenjb, 2020-10-14:
Here is a quick fix: https://github.com/intel/media-driver/pull/1064

Caution: I didn't do any in-depth analysis or testing, in fact, I only added the code for handing for specific 420 formats. Probably similar things needed for other formats too. But I hope this will give an idea of what is going wrong. Please keep in mind that all middleware applications won't set surface hint usage attributes.

XinfengZhang, 2020-10-14:
I dont think aligning with 8 is a solution, we should only align with 2 for i420/NV12.  if display region is out of aligned width, there will be green line. align with 8 just just expand the VPP region. I still suppose it is related with display region.   @FurongZhang  @Jister  could you help to double confirm

sreerenjb, 2020-10-14:
@XinfengZhang  It doesn't seems to be a display region issue. Because it happens only if you try to download an I420 image from Nv12, nv12 to nv12 works fine.

Jister, 2020-10-15:
@sreerenjb from you description, the issue happens when using vaGetImage to do conversion from 321x241 NV12 to 321x241 I420?

Jister, 2020-10-15:
As I tried to make a hard code to force to convert NV12 to I420 in GetImage, I can reproduce the issue now. The reason why NV12 has no issue is that it is aligned in DdiMediaUtil_AllocateSurface but I420 is not. So I agree with the solution #1064 to make alignment for I420 for both width and height to 2.

uartie, 2020-11-19:
@jister for patch #1088, can you confirm that #1071 and #1072 don't show up again?

PR #1088 (2020-12-01): Fix odd resolution handling of I420/YV12 formats
Fix #1037
closed 2020-12-01 by intel-mediadev