intel/media-driver #1038: vaQuery of AV1 decoder context does not return correct 10 bits info

HeJunyan, 2020-08-28:
The vainfo -a return:

VAProfileAV1Profile0/VAEntrypointVLD
    VAConfigAttribRTFormat                 : VA_RT_FORMAT_YUV420
                                             VA_RT_FORMAT_YUV422
                                             VA_RT_FORMAT_RGB32
    VAConfigAttribDecSliceMode             : VA_DEC_SLICE_MODE_NORMAL
                                             VA_DEC_SLICE_MODE_BASE
    VAConfigAttribDecProcessing            : VA_DEC_PROCESSING_NONE
    VAConfigAttribMaxPictureWidth          : 16384
    VAConfigAttribMaxPictureHeight         : 16384
    VAConfigAttribProcessingRate           : VA_PROCESSING_RATE_DECODE

VAProfileAV1Profile1/VAEntrypointVLD
    VAConfigAttribRTFormat                 : VA_RT_FORMAT_YUV420
                                             VA_RT_FORMAT_YUV422
                                             VA_RT_FORMAT_RGB32
    VAConfigAttribDecSliceMode             : VA_DEC_SLICE_MODE_NORMAL
                                             VA_DEC_SLICE_MODE_BASE
    VAConfigAttribDecProcessing            : VA_DEC_PROCESSING_NONE
    VAConfigAttribMaxPictureWidth          : 16384
    VAConfigAttribMaxPictureHeight         : 16384
    VAConfigAttribProcessingRate           : VA_PROCESSING_RATE_DECODE

And the spec says:
     seq_profile  Bit depth  Monochrome support  Chroma subsampling
     0            8 or 10    Yes                 YUV 4:2:0
     1            8 or 10    No                  YUV 4:4:4
     2            8 or 10    Yes                 YUV 4:2:2
     2            12           Yes                 YUV 4:2:0,YUV 4:2:2,YUV 4:4:4

the VAConfigAttribRTFormat of 0 profile should just contains VA_RT_FORMAT_YUV420 and VA_RT_FORMAT_YUV420_10, no 422 and RGB.
And also please make sure the return FourCC is correct.

rosetta-jpn, 2020-09-15:
The potential fix is https://github.com/intel/media-driver/pull/1044.
I am not sure gen12 supports 10bit and 12bit.
Could you mention in the pull request?

HeJunyan, 2020-09-15:
gen12 should supports 10bit, but no 12bits support. @XinfengZhang is more clear about this.

XinfengZhang, 2020-09-15:
yes, just support 10bit, actually, it should just support 8 bit 10bit 420. it is profile0, so will remove profile1 soon.

rosetta-jpn, 2020-09-15:
Added 10bits to https://github.com/intel/media-driver/pull/1044
Oh, really. Profile1 is not supported?

XinfengZhang, 2020-09-23:
Profile1 will be removed because less of  444 support.

rosetta-jpn, 2020-10-06:
Shall we close this issue because https://github.com/intel/media-driver/pull/1044 is merged?

closed 2020-10-10 by XinfengZhang