intel/media-driver #1045: Huge difference of transcoding performance between Ubuntu 16 and Windows10

tvunetworks-x, 2020-09-15:
We run this sample on Linux and Windows10   
  ./sample_multi_transcode -ec::p010 -i::h265 /home/tgl/source/tm1000v3_2160p5994_20mbps_yuv420p10bit.h265 -o::h265 ./xxx.h265 -b 30000 -async 3  -hw -n 600 -gop_size 60 -u speed

The CPU is i9-9900k. We run it on both CentOS and Ubuntu 16 Linux to compare with win10.

On Linux, it takes 17 seconds to transcode 600 frames, while on Windows it takes only 8 seconds.
Theoretically, the speeds should be same.  But it looks that it cannot fully make use of hardware acceleration on Linux.  We can reproduce it both on CentOS7.8 and Ubuntu 16.

Please explain. The screen snapshot pictures are attached.

[image]

[image]

Xiaogangli-intel, 2020-09-17:
@tvunetworks-x Could you provide the version of media driver, and also the version of KMD?

tvunetworks-x, 2020-09-17:
The MediaSDK version is 19.4 and KDM version is 4.13.

Xiaogangli-intel, 2020-09-17:
Could you provide the commit id of the media-driver which you are using?

tvunetworks-x, 2020-09-17:
Not understand what media-driver is and where to find the commit id.

We use the official release 19.4 and get the package from your GitHub
site.  You should be able to trace back the source code.

Best

-- 

*James Liu*

VP of Engineering, Shanghai Office

*TVU networks*

Suite 732-735, Asionics Hi-Tech Center
1279 Zhong Shan Road West, Building 6
Shanghai, P.R.China 200051

jamesliu@tvunetworks.com <howardhong@tvunetworks.com>

Office: +86 (21) 3209.5581

Fax: +86 (21) 3209.5515

 *Making every transmission a success*

www.tvunetworks.com <http://www.tvupack.com/>

facebook.com/TVUPack | twitter.com/tvupack | blog.tvupack.com

Xiaogangli-intel, 2020-09-28:
could you go into media-driver folder and get git log? I need to confirm which version you are using, thank you.

tvunetworks-x, 2020-09-28:
Is this what you are looking for?

-- 

*James Liu*

VP of Engineering, Shanghai Office

*TVU networks*

Suite 732-735, Asionics Hi-Tech Center
1279 Zhong Shan Road West, Building 6
Shanghai, P.R.China 200051

jamesliu@tvunetworks.com <howardhong@tvunetworks.com>

Office: +86 (21) 3209.5581

Fax: +86 (21) 3209.5515

 *Making every transmission a success*

www.tvunetworks.com <http://www.tvupack.com/>

facebook.com/TVUPack | twitter.com/tvupack | blog.tvupack.com

tvunetworks-x, 2020-09-28:
https://github.com/intel/media-driver/releases/tag/intel-media-20.1.1

we got the driver from above..

-- 

*James Liu*

VP of Engineering, Shanghai Office

*TVU networks*

Suite 732-735, Asionics Hi-Tech Center
1279 Zhong Shan Road West, Building 6
Shanghai, P.R.China 200051

jamesliu@tvunetworks.com <howardhong@tvunetworks.com>

Office: +86 (21) 3209.5581

Fax: +86 (21) 3209.5515

 *Making every transmission a success*

www.tvunetworks.com <http://www.tvupack.com/>

facebook.com/TVUPack | twitter.com/tvupack | blog.tvupack.com

Xiaogangli-intel, 2020-09-29:
Thank you. This one should be released several months ago, there are some optimizations recently, could you download the latest media driver, build locally, and try it? Or you can provide the reproduce environment, then I will try it on your machine.

tvunetworks-x, 2020-09-29:
We tried with the new driver.  The performance is around the same. Simon
will provide your environment.

On windows, it can decode 4k 10bit  more than 240fps.  While on Linux it
can only do at 130fps.

-- 

*James Liu*

VP of Engineering, Shanghai Office

*TVU networks*

Suite 732-735, Asionics Hi-Tech Center
1279 Zhong Shan Road West, Building 6
Shanghai, P.R.China 200051

jamesliu@tvunetworks.com <howardhong@tvunetworks.com>

Office: +86 (21) 3209.5581

Fax: +86 (21) 3209.5515

 *Making every transmission a success*

www.tvunetworks.com <http://www.tvupack.com/>

facebook.com/TVUPack | twitter.com/tvupack | blog.tvupack.com

tvunetworks-x, 2020-10-20:
Adding **-gpucopy::on** in command line can fix this issue. The performance on Linux is on par with Win10 now.

closed 2020-10-20 by tvunetworks-x