intel/media-driver #1047: The attribute vaCreateSurfaces query of VASurfaceAttribPixelFormat  for VPP lose some formats.

HeJunyan, 2020-09-16:
It only returns "NV12, I420, YV12, YUY2, ABGR, ARGB, xBGR, P010_10LE, BGRA, RGBA, BGR10A2_LE", some format such as VUYA is missing.

HeJunyan, 2020-09-16:
Accord to @XinfengZhang and @xhaihao, this should contain all formats of input and output of VPP.

XinfengZhang, 2020-09-16:
@FurongZhang , could you help to check it?

MicroYY, 2020-12-22:
Hi @HeJunyan 
Fix check in done. https://github.com/intel/media-driver/commit/026924fd08570b372333caac97a190ed8aa78216

XinfengZhang, 2021-01-08:
@HeJunyan , could you help to check the patch. if no other problem, please help to close it

XinfengZhang, 2021-01-29:
@HeJunyan , it should already been fixed, if you still need to add more format, or , some format is still missed. please re-open it and tell the format. thanks

closed 2021-01-29 by XinfengZhang