intel/media-driver #1051: QuerySurfaceAttributes return incompleted VASurfaceAttribMemoryType.

HeJunyan, 2020-09-17:
The query of vaQuerySurfaceAttributes return incompleted for most of the platforms and entrypoint.

1. On pre Gen12 platform, only VA_SURFACE_ATTRIB_MEM_TYPE_DRM_PRIME is returned, no VA_SURFACE_ATTRIB_MEM_TYPE_DRM_PRIME_2, which should be available according to @xhaihao.

2. On Gen12 platform, only VPP return VA_SURFACE_ATTRIB_MEM_TYPE_DRM_PRIME|VA_SURFACE_ATTRIB_MEM_TYPE_DRM_PRIME 2, other decoder/encoder even return non of them.

Please make sure the VASurfaceAttribMemoryType is correct for all platforms and all entrypoints, It is important for DMA path selection.

XinfengZhang, 2020-09-23:
VPP could accept VA_SURFACE_ATTRIB_MEM_TYPE_DRM_PRIME and VA_SURFACE_ATTRIB_MEM_TYPE_DRM_PRIME2. encode / decode should have special requirement , such as tiling.

XinfengZhang, 2020-10-25:
1. only has PRIME on pre-gen12 platform
2. like https://github.com/intel/media-driver/issues/1051#issuecomment-697169058

closed 2020-10-25 by XinfengZhang

HeJunyan, 2020-11-10:
Please reopen it.

Even the VA_SURFACE_ATTRIB_MEM_TYPE_VA is also missing.
And I think we also need this  VA_SURFACE_ATTRIB_MEM_TYPE_DRM_PRIME2 flag to support DMA mode for decoder/encoder.
We will handle the tiling by modifier. According to my opinion,  the driver should give this VA_SURFACE_ATTRIB_MEM_TYPE_DRM_PRIME2 flag for decoder context, and then we can create DMA path. Then, the modifier can be got by vaExportSurfaceHandle.

HeJunyan, 2020-11-10:
According to my understanding, the decoder can also support DMA output, and the mem type. such as tiling/compressed can be acquired by vaExportSurfaceHandle's modifier. We can transport this modifier with this DMA buffer, and let the user access the memory correctly. 

But lack of this flag, we can not enable the DMA path totally.