intel/media-driver #1056: VP9 encoding on i7-10710U with UHD Graphics for 10th Gen Intel® Processors

ajgnet, 2020-09-24:
Hi everyone — Could anyone suggest what driver I should be using to enable hardware-accelerated VP9 encoding on a **Comet Lake U-Series i7-10710U**?

### Current setup:

- iGPU: UHD Graphics on an i7-10710U
- OS: Debian Bullseye / testing 11
- Kernel: 5.8.0-1-amd64
- intel-media-va-driver-non-free: 20.2.0+ds1-1
- libva: 2.9.0-1
- vainfo: 2.8.0+ds1-1 (if at all relevant)
- libva-drm2: 2.9.0-1

Hardware support: Looking at [Encode and Decode Capabilities for 7th Generation Intel® Core™ Processors](https://software.intel.com/content/www/xl/es/develop/articles/encode-and-decode-capabilities-for-7th-generation-intel-core-processors-and-newer.html?countrylabel=Mexico) and Newer, the iGPU supports VP9 encoding.

Below is the full output of vainfo and a partial vainfo -a for the available VP9 decoder profile/entrypoint pair.

### The problem:

`# vainfo
error: XDG_RUNTIME_DIR not set in the environment.
error: can't connect to X server!
libva info: VA-API version 1.9.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_8
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.9 (libva 2.8.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 20.2.0 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD`

`VAProfileVP9Profile0/VAEntrypointVLD
    VAConfigAttribRTFormat                 : VA_RT_FORMAT_YUV420
                                             VA_RT_FORMAT_YUV422
                                             VA_RT_FORMAT_YUV420_10
                                             VA_RT_FORMAT_RGB32
                                             VA_RT_FORMAT_YUV420_10BPP
    VAConfigAttribDecSliceMode             : VA_DEC_SLICE_MODE_NORMAL
                                             VA_DEC_SLICE_MODE_BASE
    VAConfigAttribDecProcessing            : VA_DEC_PROCESSING_NONE
    VAConfigAttribMaxPictureWidth          : 8192
    VAConfigAttribMaxPictureHeight         : 8192
    VAConfigAttribProcessingRate           : VA_PROCESSING_RATE_DECODE

VAProfileVP9Profile2/VAEntrypointVLD
    VAConfigAttribRTFormat                 : VA_RT_FORMAT_YUV420
                                             VA_RT_FORMAT_YUV422
                                             VA_RT_FORMAT_YUV420_10
                                             VA_RT_FORMAT_RGB32
                                             VA_RT_FORMAT_YUV420_10BPP
    VAConfigAttribDecSliceMode             : VA_DEC_SLICE_MODE_NORMAL
                                             VA_DEC_SLICE_MODE_BASE
    VAConfigAttribDecProcessing            : VA_DEC_PROCESSING_NONE
    VAConfigAttribMaxPictureWidth          : 8192
    VAConfigAttribMaxPictureHeight         : 8192
    VAConfigAttribProcessingRate           : VA_PROCESSING_RATE_DECODE`

So, what driver should I use to enable hw-accelerated VP9 encoding if intel-vaapi-driver does not support this processor and only the VP9 decoder is available in media-driver for this platform? Any suggestions are welcome, thank you.

XinfengZhang, 2020-10-10:
hi @ajgnet , thanks for reporting the issue. sorry for late response. 2 solutions: 1. if you want to use intel-vaapi-driver, you could add pciid into https://github.com/intel/intel-vaapi-driver/blob/master/src/i965_pciids.h  2. if you want to use media driver, suppose you need to cherry-pick #717

closed 2020-10-25 by XinfengZhang

pe1uca, 2024-05-30:
@ajgnet did any of those solved the issue? Or did you do something else?