intel/media-driver #1057: [libyami][hevc encode][assert failed]

huliang4github, 2020-09-25:
./yamiencode -i /tmp/_3840x2160.NV12 -W 3840 -H 2160 -s NV12 -o /tmp/out.265
libva info: VA-API version 1.6.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /opt/intel/mediasdk/lib64/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_6
libva info: va_openDriver() returns 0
yamiencode: vaapiencoder_hevc.cpp:1567: bool YamiMediaCodec::VaapiEncoderHEVC::addPackedSliceHeader(const PicturePtr&, const VAEncSliceParameterBufferHEVC*, uint32_t) const: Assertion `0 && (!m_seqParam->seq_fields.bits.separate_colour_plane_flag)' failed.
Aborted (core dumped)

but it work well with i965 driver

./yamiencode -i /tmp/_3840x2160.NV12  -W 3840 -H 2160 -s NV12 -o /tmp/out.265
libva info: VA-API version 1.1.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_1
libva info: va_openDriver() returns 0
encode done

Xiaogangli-intel, 2020-09-29:
Cold you provide more detail about this issue? Log, platform, and reproducer. Thank you.

huliang4github, 2020-09-29:
I test it on my laptop with Intel(R) Celeron(R) CPU 3865U @ 1.80GHz , installed Ubuntu 18.04 OS;

first of all, I test hevc encode with the built-in i965_dri.so , it work well ,

later, I switch to use the iHD_drv_video.so (ver 19.3.1), and  it  failed sometimes

you can ref the log
[hevc.log.txt](https://github.com/intel/media-driver/files/5296418/hevc.log.txt)

by the way , I also test it on Intel(R) Celeron(R) CPU 3867U @ 1.80GHz, it have the same issue;

huliang4github, 2020-09-30:
the log says that it failed at libyami/encoder/vaapiencoder_hevc.cpp:1567

ASSERT(!m_seqParam->seq_fields.bits.separate_colour_plane_flag);

and I just comment this line, and it work, both with i965 and iHD.
so, per my understanding, it's the issue of libyami, 
thanks

closed 2020-09-30 by huliang4github

Xiaogangli-intel, 2020-10-09:
@huliang4github, thank  you.