intel/media-driver #1058: DRM_IOCTL_I915_GEM_APERTURE failed: No such file or directory

Feqzz, 2020-09-27:
I've been trying to get the iHD_drv_video.so to work. I have an AMD GPU which is somehow set to default when typing `$ vainfo`.

I get the error when writing:
```
$ export LIBVA_DRIVER_NAME=iHD
$ vainfo
libva info: VA-API version 1.9.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_9
DRM_IOCTL_I915_GEM_APERTURE failed: No such file or directory
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 18
vaInitialize failed with error code 18 (invalid parameter),exit
```

Although it seems to work with this input:
```
$ vainfo --display drm --device /dev/dri/renderD128
libva info: VA-API version 1.9.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_9
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.9 (libva 2.9.0.pre1)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 20.3.pre (17705613)
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD

```

XinfengZhang, 2020-10-10:
hi , suppose you have two GPU, one is from AMD, another is from Intel. you specify the GPU in the second command line. could you run "ls /dev/dri/" and show the result?

Feqzz, 2020-10-10:
@XinfengZhang 
```
$ ls /dev/dri
by-path  card0  card1  renderD128  renderD129
```

XinfengZhang, 2020-10-25:
it is strange, it should load renderD128 or card0 by default for DRM, it should be same with --device /dev/dri/renderD128, but it is not. 
it maybe because it will failed with X11 or Wayland path?

Feqzz, 2020-10-25:
@XinfengZhang 
You're maybe onto something here. I tried these different environments:

- GNOME
- GNOME on Xorg
- Openbox
- Ubuntu
- Ubuntu on Wayland

GNOME on Xorg and Ubuntu gave the same results as the original post. The others gave me this error by simply writing `$ vainfo`:
```
libva info: VA-API version 1.9.0
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
vaInitialize failed with error code -1 (unknown libva error),exit
```

In all cases `$ vainfo --display drm --device /dev/dri/renderD128` worked.

dvrogozh, 2020-10-25:
@XinfengZhang : there is a trick in the libva-drm code:
https://github.com/intel/libva/blob/e52d9de9e24cddc54f8107020fefc3f996624687/va/drm/va_drm_auth.c#L37

It means, that if user is under active X environment and tries to access libva-drm backend, then first libva will try to get authentication from the X server. And as far as I see in this path:
https://github.com/intel/libva/blob/e52d9de9e24cddc54f8107020fefc3f996624687/va/drm/va_drm_auth_x11.c#L157
actual FD you've supplied from open("/dev/dri/renderD128") is just ignored. So, if X server is connected to /dev/dri/card1 which is AMD card, then we will get authentication for the AMD card while we wanted to access intel one.

From my perspective this might be libva-drm backend bug, not media-driver bug.

manetelpian, 2020-11-11:
I am having a similar issue when I attempt to run an OpenVINO Model, and run the following command: ./text_detection_demo -m_tr /opt/intel/openvino_2021/deployment_tools/open_model_zoo/tools/downloader/intel/text-recognition-0012/FP32/text-recognition-0012.xml -i /dev/video0 -r

libva info: VA-API version 1.6.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /opt/intel/mediasdk/lib64/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_6
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
libva error: /opt/intel/mediasdk/lib64/iHD_drv_video.so init failed
libva info: va_openDriver() returns 18
MFX: Can't initialize session
[ ERROR ] Can't set the frame to begin with

wumingdabei, 2021-07-20:
> I am having a similar issue when I attempt to run an OpenVINO Model, and run the following command: ./text_detection_demo -m_tr [...]

Have you solve this problem??