intel/media-driver #1059: [libva][jpeg decode][iHD driver][ failed]

huliang4github, 2020-09-29:
I use the loadjpeg in libva-utils to decode the jpeg files, it ok with i965 driver, but have problem with iHD,  
the detail info is in the following file,
[libva-utils_loadjpeg_failed.log](https://github.com/intel/media-driver/files/5298319/libva-utils_loadjpeg_failed.log)

does any body help me? thanks so much!

XinfengZhang, 2020-10-10:
the issue is: iHD could decode jpeg , but can't display  decoded output?

huliang4github, 2020-10-10:
yes, but to be exact, it hapend with jpeg which contian the Exif info

huliang4github, 2020-10-14:
I modified tinyjpeg.c to dump decode output to file,  and use it to deocde yuv444 jpeg, the dumped file is  right , both with iHD and i965;  and using iHD the dump file is the same with using i965; but  i965  can display the yuv444p jpeg to x11 window by vaPutSurface, but iHD can't!

XinfengZhang, 2020-10-19:
it maybe related with vaPutSurface,   the implementation of vaPutSurface is totally different, and there are different VPP call in these two driver. so. thanks for reporting, it may take some time to investigate the issue.  the position of media-driver is codec/vpp driver. so we care more about codec/vpp feature.

huliang4github, 2020-10-27:
by the way, on my hw platform, decoding jpeg with i965 driver, performance can reach 300+ fps with 1080p(IMC3 format); using iHD driver only reach 100+ fps with the same jpeg.