intel/media-driver #1063: High & Main profile output has issues with old hardware decoders

chetan-prime, 2020-10-10:
My use case consists of transcoding with ffmpeg various live sources like MPEGTS cameras, etc then sending the ouput via DVB-C.

The stream comes in as mpegts, I use something similar to below command -
`ffmpeg -fflags +nobuffer -hwaccel vaapi -hwaccel_output_format vaapi  -i http://camera.ip/mpegts -f mpegts -c:v h264_vaapi -profile:v high -rc_mode:v VBR -level:v 4.1 -b:v:0 1600k -compression_level:v:0 1 -global_quality:v:0 25 -maxrate:v:0 1800k -g:v 30 -bufsize:v 3200k -c:a mp2 -b 128k http://my.mpegts.server/publish/stream`

Source & output are both 1280x720p.
I have tried all Intel media-driver versions from 18.2 upto the latest.

Output is fine via http mpegts on VLC 3.0 and all other software plays.
The mpegts stream is then converted into DVB-C output via hardware cards and sent via cable to various TV's.

Issue:
Any TV older than 2010 shows a BLANK picture on screen and can only play back audio.
One test TV details - 
Samsung LCD TV
Type: UEC32C6000
Model: UE32C6000RWXZG
Version: SQ02

Solution:
1. Change h264_vaapi profile to "constrained_baseline". Then old TV's are are able to play back the output, but as expected the output quality isn't good.
2. Change codec to lib264 - use profiles high,main or baseline & the same old TV plays back high or main profile output from libx264 just fine.
3. Change to another transcoder that uses ffmpeg + QSV. On the same hardware it works fine then using hardware decoding whatever the profile - high, main or baseline.

Points 2 & 3 indicate the output from intel media-driver for both Main & High profiles is somehow defective or has compatibility issues with older decoders. 
Can we expect the cause of the issue to be identified & fixed ?
Right now I am forced to use QSV or software encoding if I want to support high profile on old hardware decoders.

XinfengZhang, 2020-10-13:
thanks reporting @chetan-prime , but TBH, the issue is a difficult for us to debug. it seems a conformance issue or spec upgrade issue? do you have the encoded bitstream and have you compared the NAL such as SPS PPS etc. ?

chetan-prime, 2020-10-13:
I haven't yet checked the NAL, SPS etc.
But what I do know is that any source works fine on the old TV's if transcoded to "Constrained Baseline" profile using ffmpeg h264_vaapi. The issue comes only if I try to transcode to High or Main profile.
I have tried in multiple versions of the intel media driver, different versions of ffmpeg, different versions of ubuntu. The only constant is that no matter what, High or Main profile output from h264_vaapi doesn't work.

But if I use the same source, same PC, same version of ffmpeg & libx264 (i.e software encoding) the output works fine on the Old TV even in High or Main profile.

Also If I use another transcoder on the same PC that works with QSV mode, it also works fine in High or Main profile on the old TV. For some reason using the old QSV mode gives spec-compliant output for high profile like libx264.

Do you recommend any tool to compare the NAL, SPS etc. ?

XinfengZhang, 2020-10-21:
libva trace could be used to compare the parameters

chetan-prime, 2020-10-21:
Thanks @XinfengZhang , I have dumped and uploaded the logs with this comment.

First is High profile transcode using h264_vaapi. As you can see in below picture the TV detects it as an audio stream only.
[libva-high.zip](https://github.com/intel/media-driver/files/5415046/libva-high.zip)
[image]

Next I transcode the same source with h264_vaapi and Constrained Baseline profile. As you can see with below photo the TV now plays back the stream just fine.
[libva-constrained.zip](https://github.com/intel/media-driver/files/5415031/libva-constrained.zip)
[image]

I could send debug logs for libx264 too if it will be useful to compare. With lib264 the same TV is able to play back the same stream with "High" profile. Somehow libx264 is able to generate a stream that is more compatible with all decoders.

chetan-prime, 2021-02-26:
Thanks for moving this to medium priority.
Kindly inform if there are any changes to the master branch and if I should do any testing on older TV once again with the latest drivers

Xiaogangli-intel, 2022-04-19:
Hi @chetan-prime, sorry for later response for this issue. Did you tried the latest driver? you can always try the master branch.

Xiaogangli-intel, 2022-04-25:
Hi @chetan-prime, do you still have any comments on this issue? if haven't, I will close this issue.

chetan-prime, 2022-04-25:
Hi @Xiaogangli-intel , the person with hardware on his end is unresponsive or unwilling to test.
I will close the issue for now & reopen if he still finds an issue in further testing

closed 2022-04-25 by chetan-prime