intel/media-driver #1066: [libyami][jpeg decode][iHD driver][ failed]

huliang4github, 2020-10-14:
use yamidecode in  libyami-utils to decode jpeg,  and dump to file

./yamidecode -i /tmp/1920x1080.jpeg -m 0 -w 1 -o /tmp/

i965 driver work well, while switch to use iHD driver,  the size of the dump out file is right, but content is wrong!

XinfengZhang, 2020-10-19:
is it same as #1059 ?

huliang4github, 2020-10-23:
it seem not completely the same.  
#1059 :  libva can decode all format jpegs, just can't display the yuv444p jpeg to X window via vaPutSurface;
#1066: libyami can't decode all the format jpegs, I  dump the  "frame" and "dest"  in decodeoutput.cpp like following code snippets
 
bool DecodeOutputDump::output(const SharedPtr<VideoFrame>& frame)
{
    if (!initOutput(frame))
        return false;
    SharedPtr<VideoFrame> dest = m_convert->convert(frame);
    return m_output->output(dest);
}

both output content is incorrect!

XinfengZhang, 2020-10-25:
only libyami has this issue?