intel/media-driver #1067: build fails with -Werror on usage of deprecated VAAPI fields

dvrogozh, 2020-10-19:
Commits:
* https://github.com/intel/libva/commit/1c5de1eb22007a5816cd831745e3c1b169bf34d0
* https://github.com/intel/media-driver/commit/9881f7ae90665106f0e5a34a81b31aa774a27b25

```
/docker/worker/driver/build_dir/repos/media-driver/media_driver/linux/gen12/codec/ddi/media_ddi_decode_av1_g12.cpp:80:62: error: ‘_VASliceParameterBufferAV1::tg_start’ is deprecated [-Werror=deprecated-declarations]
         tileParams->m_startTileIdx              = pTileCtrl->tg_start;
                                                              ^~~~~~~~
In file included from /docker/worker/driver/build_dir/dependencies/libva/usr/local/include/va/va.h:4779,
                 from /docker/worker/driver/build_dir/repos/media-driver/media_driver/linux/common/ddi/media_libva.h:30,
                 from /docker/worker/driver/build_dir/repos/media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.h:30,
                 from /docker/worker/driver/build_dir/repos/media-driver/media_driver/linux/gen12/codec/ddi/media_ddi_decode_av1_g12.cpp:27:
/docker/worker/driver/build_dir/dependencies/libva/usr/local/include/va/va_dec_av1.h:637:29: note: declared here
     va_deprecated uint16_t  tg_start;
                             ^~~~~~~~
```

uartie, 2020-10-19:
Feel free to use my temporary workaround if this can't be properly fixed right away.  https://github.com/uartie/media-driver/commit/fe25ea249127062b30e94f9bf512bd505f600d22

XinfengZhang, 2020-10-21:
should already been fixed by https://github.com/intel/media-driver/commit/250db2198aa6a6024fe20c1f18f9ebbf1939b5c8

closed 2020-10-21 by XinfengZhang