intel/media-driver #1071: [regression][gstreamer] VPP rotation 90/270 YV12 corruption

uartie, 2020-10-28:
Since commit 450fcc06c0eb4a61a2f5f73b3e07e0e875a484d6 (#1064):

```
commit 450fcc06c0eb4a61a2f5f73b3e07e0e875a484d6
Author: Sreerenj Balachandran <sreerenj.balachandran@intel.com>
Date:   Tue Oct 13 18:38:00 2020 -0700

    media_libva_util: Fix odd resolution handling of yuv420 formats
    
    Ensure 2-byte width & height alignment for I420/YV12/IYUV surfaces
    when there is no usage hint set for Encode or VPP write operations.
    It seems the  odd resolution surface post-processing operation
    (eg: nv12 to i420 conversion via vaGetImage API), causing a green
    column on the right side of the video without this alignment.
    
    Fixing #1037
```

YV12 rotation at 90 or 270 degress with vaapipostproc and msdkvpp produces corrupted output.

```
gst-launch-1.0 -vf videotestsrc num-buffers=1 ! video/x-raw,format=YV12,width=1920,height=1080 \
  ! vaapipostproc video-direction=90r ! video/x-raw,format=YV12 \
  ! checksumsink2 file-checksum=false qos=false frame-checksum=false \
  plane-checksum=false dump-output=true dump-location=gst-vaapi-1080x1920_YV12.yuv

gst-launch-1.0 -vf videotestsrc num-buffers=1 ! video/x-raw,format=YV12,width=1920,height=1080 \
  ! msdkvpp video-direction=90r ! video/x-raw,format=YV12 \
  ! checksumsink2 file-checksum=false qos=false frame-checksum=false \
  plane-checksum=false dump-output=true dump-location=gst-msdk-1080x1920_YV12.yuv
```

Actual Result:
<p>[image]</p>

Expected Result:
<p>[image]</p>

cc: @sreerenjb

sreerenjb, 2020-10-28:
Ouch! we should revert immediately then.
@Jister @XinfengZhang 

Quickly tested with xvimagesink but didn't see any issue. Probably the conversion involved in the middle is hiding the actual issue.
Originally I proposed 8 byte alignment to match with i965.  Any chance to test with 
--- a/media_driver/linux/common/ddi/media_libva_util.cpp
+++ b/media_driver/linux/common/ddi/media_libva_util.cpp
@@ -208,8 +208,8 @@ VAStatus DdiMediaUtil_AllocateSurface(
                 !(mediaSurface->surfaceUsageHint & VA_SURFACE_ATTRIB_USAGE_HINT_VPP_WRITE))
             {
                 tileformat = I915_TILING_NONE;
-                alignedWidth = MOS_ALIGN_CEIL(width, 2);
-                alignedHeight = MOS_ALIGN_CEIL(height, 2);
+                alignedWidth = MOS_ALIGN_CEIL(width, 8);
+                alignedHeight = MOS_ALIGN_CEIL(height, 8);
                 break;
             }

sreerenjb, 2020-10-28:
Aha, formatting issue.
Change **alignedWidth = MOS_ALIGN_CEIL(width, 2);  to  alignedWidth = MOS_ALIGN_CEIL(width, 8);**
and  **alignedHeight = MOS_ALIGN_CEIL(height, 2); to  alignedHeight = MOS_ALIGN_CEIL(height, 8);**

sreerenjb, 2020-10-28:
@uartie 
I tested with 
gst-launch-1.0 -vf videotestsrc num-buffers=1 ! video/x-raw,format=YV12,width=1920,height=1080   ! vaapipostproc video-direction=90r ! video/x-raw,format=YV12   ! checksumsink2 file-checksum=false qos=false frame-checksum=false   plane-checksum=false dump-output=true dump-location=gst-vaapi-1080x1920_YV12.yuv

and play with ffmpeg:
ffplay -f rawvideo -pixel_format yuv420p -video_size 1080x1920 gst-vaapi-1080x1920_YV12.yuv

**Not seeing any corruption.** But since ffmpeg doesn't have any YV12 option , I provided yuv420p. So color mismatch is there in chroma.

uartie, 2020-10-28:
> @uartie
> I tested with
> gst-launch-1.0 -vf videotestsrc num-buffers=1 ! video/x-raw,format=YV12,width=1920,height=1080 ! vaapipostproc video-direction=90r ! video/x-raw,format=YV12 ! checksumsink2 [...]

You mean you're not seeing any corruption *with* your aforementioned changes or *without* your changes?

sreerenjb, 2020-10-29:
Without any changes. I mean, use the iHD master which has https://github.com/intel/media-driver/commit/450fcc06c0eb4a61a2f5f73b3e07e0e875a484d6

uartie, 2020-10-29:
@sreerenjb I just tried with your change in https://github.com/intel/media-driver/issues/1071#issuecomment-718233128.  But it did not seem to fix the issue.

@XinfengZhang @Sherry-Lin can you please revert 450fcc06c0eb4a61a2f5f73b3e07e0e875a484d6 commit.  I'm seeing more use-cases failing in our full testing too.

XinfengZhang, 2020-10-29:
sure， will revert it

sreerenjb, 2020-10-29:
Ya, revert and investigate is the option :)

uartie, 2020-10-30:
reverted in ca794cb70bdd5faf134324310b7054f240dd77f9 and fixes this issue.

closed 2020-10-30 by uartie