intel/media-driver #1073: media-driver-intel-media-19.4.0r build failed, no MhwBltInterface

YaboWei, 2020-10-29:
[  2%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/media_interfaces/media_interfaces.cpp.o
In file included from /home/web_server/media-driver-intel-media-19.4.0r/media_driver/agnostic/common/media_interfaces/media_interfaces.h:31:0,
                 from /home/web_server/media-driver-intel-media-19.4.0r/media_driver/agnostic/common/media_interfaces/media_interfaces_mhw.h:30,
                 from /home/web_server/media-driver-intel-media-19.4.0r/media_driver/linux/common/media_interfaces/media_interfaces.cpp:27:
/home/web_server/media-driver-intel-media-19.4.0r/media_driver/agnostic/common/os/mos_utilities.h: In instantiation of 'void MOS_DeleteUtil(_Ty&) [with _Ty = MhwBltInterface*]':
/home/web_server/media-driver-intel-media-19.4.0r/media_driver/linux/common/media_interfaces/media_interfaces.cpp:201:5:   required from here
/home/web_server/media-driver-intel-media-19.4.0r/media_driver/agnostic/common/os/mos_utilities.h:249:9: error: possible problem detected in invocation of delete operator: [-Werror]
         delete(ptr);
         ^
/home/web_server/media-driver-intel-media-19.4.0r/media_driver/agnostic/common/os/mos_utilities.h:249:9: error: invalid use of incomplete type 'class MhwBltInterface' [-Werror]
In file included from /home/web_server/media-driver-intel-media-19.4.0r/media_driver/linux/common/media_interfaces/media_interfaces.cpp:27:0:
/home/web_server/media-driver-intel-media-19.4.0r/media_driver/agnostic/common/media_interfaces/media_interfaces_mhw.h:47:7: error: forward declaration of 'class MhwBltInterface' [-Werror]
 class MhwBltInterface;
       ^
In file included from /home/web_server/media-driver-intel-media-19.4.0r/media_driver/agnostic/common/media_interfaces/media_interfaces.h:31:0,
                 from /home/web_server/media-driver-intel-media-19.4.0r/media_driver/agnostic/common/media_interfaces/media_interfaces_mhw.h:30,
                 from /home/web_server/media-driver-intel-media-19.4.0r/media_driver/linux/common/media_interfaces/media_interfaces.cpp:27:
/home/web_server/media-driver-intel-media-19.4.0r/media_driver/agnostic/common/os/mos_utilities.h:249:9: note: neither the destructor nor the class-specific operator delete will be called, even if they are declared when the class is defined
         delete(ptr);
         ^
cc1plus: error: unrecognized command line option "-Wno-delete-incomplete" [-Werror]
cc1plus: all warnings being treated as errors
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/media_interfaces/media_interfaces.cpp.o] Error 1
make[1]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
make: *** [all] Error 2

XinfengZhang, 2020-11-02:
could you help to figure out  your compiler , version and env.  and more details.

XinfengZhang, 2021-03-09:
closed, suppose it is related with environment .  with gcc 7.5.0 and cmake 3.10.2, it could be built successfully.