intel/media-driver #1074: Encode: VP9: Failed to create the correct VP8 frame header for 4:4:4 profile.

HeJunyan, 2020-10-30:
When the 4:4:4 chrome is seleced, the sub_sampling_x/y is not very correct in VP9 frame header.

I notice:

In  linux/common/codec/ddi/media_libvpx_vp9.cpp:

96     if ((codecProfile == VP9_PROFILE_1) ||
97         (codecProfile == VP9_PROFILE_3))
98     {
99         /* sub_sampling_x/y */
100         /* Currently the sub_sampling_x = 0, sub_sampling_y = 1 */
101         vp9_wb_write_bit(wb, 0);
102         vp9_wb_write_bit(wb, 0);
103         vp9_wb_write_bit(wb, 0); // unused
104     }
105 }

Seems not very correct.

XinfengZhang, 2020-11-02:
thanks for reporting.  most usage  application should already packed it, and headerInsertFlag of
https://github.com/intel/media-driver/blob/master/media_driver/linux/common/codec/ddi/media_ddi_encode_vp9.cpp#L269
is true.  and in this part, it only know the profile ,still not know the chrome sub_sampling.

SiewHoon, 2020-11-09:
Any update on this? The fix need to be in media-driver or gst-vaapi?

XinfengZhang, 2020-11-09:
it depend on your usage case, if you pack the header in gst-vaapi and sent it to driver. the fix should be exist in gst-vaapi, if not , it should be fixed in media_driver.

SiewHoon, 2020-11-09:
@XinfengZhang , my test is run using gst-vaapi.  I don't have any app to run it beside gst-vaapi. 

On libva-utils component, the vp9enc.c does not AYUV and Y410 color format.  Not able to verify the issue seeing it or not. 

@HeJunyan what is your opinion at here?

PR #1109 (2020-12-23): [Encode] Fixed VP9 444 frame header
According VP9 spec section 7.2.2 color config semantics,
both subsampling_x and subsampling_y should be 0 for YUV444.
Only YUV440 subsampling_x=0 and subsampling_y=1.

fixes #1074
commit a33eaaec (2020-12-23): [Encode] Fixed VP9 444 frame header

According VP9 spec section 7.2.2 color config semantics,
both subsampling_x and subsampling_y should be 0 for YUV444.
Only YUV440 subsampling_x=0 and subsampling_y=1.

fixes #1074
closed 2020-12-23 by intel-mediadev