intel/media-driver #1075: ffmpeg-vaapi/qsv has different md5 checksum for vp9 12bit 444 decode (Y412)

FocusLuo, 2020-11-04:
ffmpeg -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_flags allow_profile_mismatch -filter_hw_device hw -v verbose -i /home/focus/workspace_focus/full/vp9/12bit/444/vp9_dec_12bit_420_4096x8192.ivf -pix_fmt y412 -f rawvideo -vsync passthrough -autoscale 0 -vframes 3 -y  test-vaapi.yuv
ffmpeg -init_hw_device qsv=qsv:hw -hwaccel qsv -filter_hw_device qsv -v verbose -c:v vp9_qsv -i /home/focus/workspace_focus/full/vp9/12bit/444/vp9_dec_12bit_420_4096x8192.ivf -vf 'hwdownload,format=y412' -pix_fmt y412 -f rawvideo -vsync passthrough -autoscale 0 -vframes 3 -y test-qsv.yuv

root@yami-tgl:/home/focus/workspace_focus/vaapi-fits-smoke# ll
-rw-r--r--  1 root root 805306368 Oct 19 15:52 test-qsv.yuv
-rw-r--r--  1 root root 805306368 Oct 19 15:52 test-vaapi.yuv

root@yami-tgl:/home/focus/workspace_focus/vaapi-fits-smoke# md5sum test-vaapi.yuv test-qsv.yuv
1878b01f200936fb23a5d6c9b3a2b61d  test-vaapi.yuv
cd5412c6e1a2d76e0d1a3c58cc60850b  test-qsv.yuv

FocusLuo, 2020-11-04:
"The issue is related to Y416, not Y412.
The driver fills 12bits for Y/U/V, but 16 bits for A when decoding 12bit stream and using Y416 as output format, so I think the driver should fill 12bits for A too."

XinfengZhang, 2020-11-05:
thanks reporting, yes, to set the alpha value to be 1, we set it to 0xffff for Y416 and 0xfff0 for Y412, it is not suitable, because the stream are both 12 bit. so , we should set it to 0xfff0.

wangyan-intel, 2020-11-17:
@FocusLuo I have tried your ffmpeg command line with your clip. But I find vp9_dec_12bit_420_4096x8192.ivf is YUV420 12-bit clip in fact.  And you uses "-pix_fmt y412" for output YUV. So it will enable VP pipeline for CSC from YUV420 12-bit to Y412 (YUV444 12 bit). And I guess qsv may use the different CSC parameter with vaapi path. I also tried VP9/HEVC YUV444 12 bit clips. The result is same between qsv and vaapi. So I think it may be VP not decoding issue. Could you please double check it?

PR #1090 (2020-11-26): Use different default alpha value for 12 bit decoding
Fixes #1075.
commit 6a221e04 (2020-11-26): Use different default alpha value for 12 bit decoding

Fixes #1075.
closed 2020-11-26 by intel-mediadev