intel/media-driver #1080: Crash in 19.4.1 on Ubuntu 18.04.5 (kernel 5.4.0-52)

chao-camect, 2020-11-08:
The stack trace is like:
_ZN18GpuContextSpecific16RegisterResourceEP22_MOS_SPECIFIC_RESOURCEb
    media-driver/media_driver/linux/common/os/mos_gpucontext_specific.cpp:328 pc=0x7f6eaa29f89d
_ZN18GpuContextSpecific16RegisterResourceEP22_MOS_SPECIFIC_RESOURCEb
    media-driver/media_driver/linux/common/os/mos_gpucontext_specific.cpp:328 pc=0x7f6ea9f954f6
_Z30Mhw_AddResourceToCmd_PatchListP14_MOS_INTERFACEP19_MOS_COMMAND_BUFFERP20_MHW_RESOURCE_PARAMS
    media-driver/media_driver/agnostic/common/hw/mhw_utilities.c:181 pc=0x7f6ea9f954f6
_Z30Mhw_AddResourceToCmd_PatchListP14_MOS_INTERFACEP19_MOS_COMMAND_BUFFERP20_MHW_RESOURCE_PARAMS
    media-driver/media_driver/agnostic/common/hw/mhw_utilities.c:181 pc=0x7f6eaa038fcb
_ZN21MhwMiInterfaceGenericI11mhw_mi_g9_XE24AddMiStoreRegisterMemCmdEP19_MOS_COMMAND_BUFFERP33_MHW_MI_STORE_REGISTER_MEM_PARAMS
    media-driver/media_driver/agnostic/common/hw/mhw_mi_generic.h:464 pc=0x7f6eaa038fcb
_ZN21MhwMiInterfaceGenericI11mhw_mi_g9_XE24AddMiStoreRegisterMemCmdEP19_MOS_COMMAND_BUFFERP33_MHW_MI_STORE_REGISTER_MEM_PARAMS
    media-driver/media_driver/agnostic/common/hw/mhw_mi_generic.h:464 pc=0x7f6ea9ee7ab4
_ZN14CodechalDecode15EndStatusReportER26CodechalDecodeStatusReportP19_MOS_COMMAND_BUFFER
    media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1443 pc=0x7f6ea9ee7ab4
_ZN14CodechalDecode15EndStatusReportER26CodechalDecodeStatusReportP19_MOS_COMMAND_BUFFER
    media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1443 pc=0x7f6ea9ef473b
_ZN17CodechalDecodeAvc20DecodePrimitiveLevelEv
    media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp:1841 pc=0x7f6ea9ef473b
_ZN17CodechalDecodeAvc20DecodePrimitiveLevelEv
    media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp:1841 pc=0x7f6ea9eea9a1
_ZN14CodechalDecode7ExecuteEPv
    media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1284 pc=0x7f6ea9eea9a1
_ZN14CodechalDecode7ExecuteEPv
    media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1284 pc=0x7f6eaa248115
_ZN14DdiMediaDecode10EndPictureEP15VADriverContextj
    media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:806 pc=0x7f6eaa248115
_ZN14DdiMediaDecode10EndPictureEP15VADriverContextj
    media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:806 pc=0x7f6eaa249b15
_Z20DdiDecode_EndPictureP15VADriverContextj
    media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:153 pc=0x7f6eaa249b15
_Z20DdiDecode_EndPictureP15VADriverContextj
    media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:153 pc=0x7f6eaa271a09
DdiMedia_EndPicture
    media-driver/media_driver/linux/common/ddi/media_libva.cpp:3368 pc=0x7f6eaa271a09
DdiMedia_EndPicture
    media-driver/media_driver/linux/common/ddi/media_libva.cpp:3368 pc=0x7f6eeaa1e5ff

chao-camect, 2020-11-08:
Another stack trace
_ZN18GpuContextSpecific13SetPatchEntryEP14_MOS_INTERFACEP23_MOS_PATCH_ENTRY_PARAMS
    media-driver/media_driver/linux/common/os/mos_gpucontext_specific.cpp:376 pc=0x7f63ec61fa01
_ZN18GpuContextSpecific13SetPatchEntryEP14_MOS_INTERFACEP23_MOS_PATCH_ENTRY_PARAMS
    media-driver/media_driver/linux/common/os/mos_gpucontext_specific.cpp:376 pc=0x7f63ec315622
_Z30Mhw_AddResourceToCmd_PatchListP14_MOS_INTERFACEP19_MOS_COMMAND_BUFFERP20_MHW_RESOURCE_PARAMS
    media-driver/media_driver/agnostic/common/hw/mhw_utilities.c:230 pc=0x7f63ec315622
_Z30Mhw_AddResourceToCmd_PatchListP14_MOS_INTERFACEP19_MOS_COMMAND_BUFFERP20_MHW_RESOURCE_PARAMS
    media-driver/media_driver/agnostic/common/hw/mhw_utilities.c:230 pc=0x7f63ec4149d9
_ZN22MhwVdboxMfxInterfaceG9I20mhw_vdbox_mfx_g9_kblE22AddMfxAvcDirectmodeCmdEP19_MOS_COMMAND_BUFFERP32_MHW_VDBOX_AVC_DIRECTMODE_PARAMS
    media-driver/media_driver/agnostic/gen9/hw/vdbox/mhw_vdbox_mfx_g9_X.h:1405 pc=0x7f63ec4149d9
_ZN22MhwVdboxMfxInterfaceG9I20mhw_vdbox_mfx_g9_kblE22AddMfxAvcDirectmodeCmdEP19_MOS_COMMAND_BUFFERP32_MHW_VDBOX_AVC_DIRECTMODE_PARAMS
    media-driver/media_driver/agnostic/gen9/hw/vdbox/mhw_vdbox_mfx_g9_X.h:1405 pc=0x7f63ec271016
_ZN17CodechalDecodeAvc14AddPictureCmdsEP19_MOS_COMMAND_BUFFERPNS_14PIC_MHW_PARAMSE
    media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp:1526 pc=0x7f63ec271016
_ZN17CodechalDecodeAvc14AddPictureCmdsEP19_MOS_COMMAND_BUFFERPNS_14PIC_MHW_PARAMSE
    media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp:1526 pc=0x7f63ec272de3
_ZN17CodechalDecodeAvc16DecodeStateLevelEv
    media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp:1570 pc=0x7f63ec272de3
_ZN17CodechalDecodeAvc16DecodeStateLevelEv
    media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp:1570 pc=0x7f63ec26a98d
_ZN14CodechalDecode7ExecuteEPv
    media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1280 pc=0x7f63ec26a98d
_ZN14CodechalDecode7ExecuteEPv
    media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1280 pc=0x7f63ec5c8115
_ZN14DdiMediaDecode10EndPictureEP15VADriverContextj
    media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:806 pc=0x7f63ec5c8115
_ZN14DdiMediaDecode10EndPictureEP15VADriverContextj
    media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:806 pc=0x7f63ec5c9b15
_Z20DdiDecode_EndPictureP15VADriverContextj
    media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:153 pc=0x7f63ec5c9b15
_Z20DdiDecode_EndPictureP15VADriverContextj
    media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:153 pc=0x7f63ec5f1a09
DdiMedia_EndPicture
    media-driver/media_driver/linux/common/ddi/media_libva.cpp:3368 pc=0x7f63ec5f1a09
DdiMedia_EndPicture
    media-driver/media_driver/linux/common/ddi/media_libva.cpp:3368 pc=0x7f642ce9e5ff

YijinLiu, 2020-11-10:
In case it matters, the CPU is i7-10700

wangyan-intel, 2020-11-10:
Could you please share your command and video clip? We can reproduce it.

YijinLiu, 2020-11-10:
Unfortunately, I cannot provide such a command. It's program that transcodes realtime stream from security cameras.
Here is another different stack trace:
<pre>
#1  0x00007fffb55809dd in CodechalDecode::Execute (this=0x17908000, params=0x2bcb6010) at media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1211
#2  0x00007fffb58de116 in DdiMediaDecode::EndPicture (this=0xd80a300, ctx=0xed1fde0, context=268435456) at media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:806
#3  0x00007fffb58dfb16 in DdiDecode_EndPicture (ctx=ctx@entry=0xed1fde0, context=context@entry=268435456) at media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:153
#4  0x00007fffb5907a0a in DdiMedia_EndPicture (ctx=0xed1fde0, context=268435456) at media-driver/media_driver/linux/common/ddi/media_libva.cpp:3368
</pre>

YijinLiu, 2020-11-10:
BTW, I am using ffmpeg-4.1.6 with libva 2.9.0

wangyan-intel, 2020-11-10:
What is your ffmpeg command?

YijinLiu, 2020-11-10:
We call the ffmpeg library as part of a complicated process. I don't think there is anything wrong the way we use ffmpeg...
The same program works very well with intel-vaapi-driver. This frequent crashes only happen when we switch to iHD driver.
It seems to me it's an index-out-of-boundary bug?

wangyan-intel, 2020-11-10:
We need one method to reproduce your issue.

wangyan-intel, 2020-11-10:
As my understanding, You are using i7-10700 + iHD driver to decode the video clip which is encoded by your camera. I guess the encoded clip may have some error which causes this issue.

YijinLiu, 2020-11-11:
I think you are probably right that there is some issue of the stream, either due to the camera or network.
If this is what happened, the driver could return some error instead of crash...
I understand you want a method to repro it. However, as you said it's likely caused by the data. Even I can give you some ffmpeg command to mimic what I am doing in the program, it won't help.
For me, the stack trace seems to have enough information to indicate what's wrong.

wangyan-intel, 2020-11-12:
Yes. The driver should return error and not crash. So the error clip may be useful to locate this issue in media driver.

Jexu, 2020-12-21:
It is hard to locate the issue from your log. The crash always happens in same function or in random function? can you share the core dump when crash happened? Then we can get the full call stack from the core file.

YijinLiu, 2020-12-25:
I am having some difficulty getting the core file for you. I'll try..
However, I posted the full stack traces (except the part of my program, which is not relevant here) earlier on. 
I don't think it's random crashes. It's not in one function either. The same binary works fine with i965 driver.
Here are two more crashes:
<pre>
mos_add_validate_buffer2
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:569 pc=0x7fe7dce08bad
mos_add_validate_buffer2
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:569 pc=0x7fe7dce0b64a
mos_gem_bo_process_reloc2
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:2488 pc=0x7fe7dce0b64a
mos_gem_bo_process_reloc2
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:2488 pc=0x7fe7dce104f3
do_exec2
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:2672 pc=0x7fe7dce104f3
do_exec2
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:2672 pc=0x7fe7dce20010
non-Go function
    pc=0x7fe7dce20010
_Z32Mos_Specific_SubmitCommandBufferP14_MOS_INTERFACEP19_MOS_COMMAND_BUFFERi
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/mos_os_specific.c:3888 pc=0x7fe7dce17e4f
_Z32Mos_Specific_SubmitCommandBufferP14_MOS_INTERFACEP19_MOS_COMMAND_BUFFERi
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/mos_os_specific.c:3888 pc=0x7fe7dca738a9
_ZN17CodechalDecodeAvc20DecodePrimitiveLevelEv
    /tmp/4shome_tmp/build/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp:1876 pc=0x7fe7dca738a9
_ZN17CodechalDecodeAvc20DecodePrimitiveLevelEv
    /tmp/4shome_tmp/build/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp:1876 pc=0x7fe7dca699a1
_ZN14CodechalDecode7ExecuteEPv
    /tmp/4shome_tmp/build/media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1284 pc=0x7fe7dca699a1
_ZN14CodechalDecode7ExecuteEPv
    /tmp/4shome_tmp/build/media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1284 pc=0x7fe7dcdc7115
_ZN14DdiMediaDecode10EndPictureEP15VADriverContextj
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:806 pc=0x7fe7dcdc7115
_ZN14DdiMediaDecode10EndPictureEP15VADriverContextj
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:806 pc=0x7fe7dcdc8b15
_Z20DdiDecode_EndPictureP15VADriverContextj
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:153 pc=0x7fe7dcdc8b15
_Z20DdiDecode_EndPictureP15VADriverContextj
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:153 pc=0x7fe7dcdf0a09
DdiMedia_EndPicture
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3368 pc=0x7fe7dcdf0a09
DdiMedia_EndPicture
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3368 pc=0x7fe81dddcf2f
non-Go function
    pc=0x7fe81dddcf2f
non-Go function
    pc=0x7fe81dab1419
non-Go function
    pc=0x7fe81dababb0
non-Go function
    pc=0x7fe81d8d0bf7
non-Go function
    pc=0x7fe81d8e6839
non-Go function
    pc=0x7fe81d85414b
non-Go function
    pc=0x7fe81d854df7
_ZN5Input11DecodeVideoEP8AVPacket
</pre>
<pre>
mos_gem_bo_process_reloc2
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:2473 pc=0x7ff67825c624
mos_gem_bo_process_reloc2
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:2473 pc=0x7ff67825c634
mos_gem_bo_process_reloc2
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:2482 pc=0x7ff67825c634
mos_gem_bo_process_reloc2
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:2482 pc=0x7ff6782614f3
do_exec2
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:2672 pc=0x7ff6782614f3
[... 22 lines cut ...]
_Z20DdiDecode_EndPictureP15VADriverContextj
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:153 pc=0x7ff678241a09
DdiMedia_EndPicture
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3368 pc=0x7ff678241a09
DdiMedia_EndPicture
    /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3368 pc=0x7ff6b937df2f
non-Go function
    pc=0x7ff6b937df2f
non-Go function
    pc=0x7ff6b9052419
non-Go function
    pc=0x7ff6b905bbb0
non-Go function
    pc=0x7ff6b8e71bf7
non-Go function
    pc=0x7ff6b8e87839
non-Go function
    pc=0x7ff6b8df514b
non-Go function
    pc=0x7ff6b8df5df7
_ZN5Input11DecodeVideoEP8AVPacket
</pre>

_ZN5Input11DecodeVideoEP8AVPacket calls ffmpeg avcodec_send_packet/avcodec_receive_frame
Somehow, it failed to print the stacktrace for ffmpeg.

YijinLiu, 2020-12-29:
This one is same as the first one, except it has some ffmpeg functions.
<pre>
#0  0x00007fffb517589d in GpuContextSpecific::RegisterResource (this=0x2a202ba0, osResource=0x282e0250, writeFlag=true) at /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/mos_gpucontext_specific.cpp:328
#1  0x00007fffb4e6b4f7 in Mhw_AddResourceToCmd_PatchList (pOsInterface=0x1ecc2000, pCmdBuffer=0x7fff7ea8a7d0, pParams=0x7fff7ea8a710) at /tmp/4shome_tmp/build/media-driver/media_driver/agnostic/common/hw/mhw_utilities.c:181
#2  0x00007fffb4f0eeac in MhwMiInterfaceGeneric<mhw_mi_g9_X>::AddMiStoreDataImmCmd (this=0x31c19040, cmdBuffer=0x7fff7ea8a7d0, params=0x7fff7ea8a7a0)
    at /tmp/4shome_tmp/build/media-driver/media_driver/agnostic/common/hw/mhw_mi_generic.h:319
#3  0x00007fffb4dbf896 in CodechalDecode::StartStatusReport (this=this@entry=0x282e0000, cmdBuffer=cmdBuffer@entry=0x7fff7ea8a7d0)
    at /tmp/4shome_tmp/build/media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1371
#4  0x00007fffb4dc8d67 in CodechalDecodeAvc::DecodeStateLevel (this=0x282e0000) at /tmp/4shome_tmp/build/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp:1567
#5  0x00007fffb4dc098e in CodechalDecode::Execute (this=0x282e0000, params=0x19c6fa10) at /tmp/4shome_tmp/build/media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1280
#6  0x00007fffb511e116 in DdiMediaDecode::EndPicture (this=0x1edec580, ctx=0x1c73ab20, context=268435456) at /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:806
#7  0x00007fffb511fb16 in DdiDecode_EndPicture (ctx=ctx@entry=0x1c73ab20, context=context@entry=268435456) at /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:153
#8  0x00007fffb5147a0a in DdiMedia_EndPicture (ctx=0x1c73ab20, context=268435456) at /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3368
#9  0x00007ffff6133f30 in vaEndPicture () from bin/libffmpeg-cam4linux.so
#10 0x00007ffff5e0841a in ff_vaapi_decode_issue () from bin/libffmpeg-cam4linux.so
#11 0x00007ffff5e11bb1 in ?? () from bin/libffmpeg-cam4linux.so
#12 0x00007ffff5c27bf8 in ff_h264_field_end () from bin/libffmpeg-cam4linux.so
#13 0x00007ffff5c3d83a in ?? () from bin/libffmpeg-cam4linux.so
#14 0x00007ffff5bab14c in ?? () from bin/libffmpeg-cam4linux.so
#15 0x00007ffff5babdf8 in avcodec_send_packet () from bin/libffmpeg-cam4linux.so
</pre>

YijinLiu, 2020-12-30:
Just took some time reading the stack traces. It seems to me caused by memory corruption. Our code calling avcodec_send_packet runs in one thread. So it's not caused by contention.
Not sure whether the following is useful. All crashes have these in common:
<pre>
linux/common/codec/ddi/media_ddi_decode_base.cpp:806
linux/common/codec/ddi/media_libva_decoder.cpp:153
linux/common/ddi/media_libva.cpp:3368
</pre>

Jexu, 2021-01-04:
[igfx_user_feature.txt](https://github.com/intel/media-driver/files/5763530/igfx_user_feature.txt)

Hi @YijinLiu 
Please use igfx_user_feature.txt to help dump the driver log:
1.  mv igfx_user_feature.txt /etc/igfx_user_feature.txt
2. mkdir -p /tmp/codechal_dump
3. run your case
then share the xxx.log in /tmp/codechal_dump with me, then i can check where is crashed in driver.

YijinLiu, 2021-01-04:
Thanks. Dropped the file.
I also upgraded the driver to intel-media-20.3.0
I'll report back if I'll find sth.

YijinLiu, 2021-01-09:
I got another crash from 20.3.0. /tmp/codechal_dump is empty though.
Stack trace is like
<pre>
#14 0x00007f9f5b3346f9 in GpuContextSpecific::RegisterResource (this=0x4a0a, osResource=0x4a34, writeFlag=6) at /tmp/4shome_tmp/build/media-driver/media_driver/linux/common/os/mos_gpucontext_specific.cpp:398
#15 0x00007f9f5b0470d4 in Mhw_AddResourceToCmd_PatchList (pOsInterface=0x0, pCmdBuffer=0x4a0a, pParams=0x0) at /tmp/4shome_tmp/build/media-driver/media_driver/agnostic/common/hw/mhw_utilities.c:187
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
</pre>

YijinLiu, 2021-01-20:
I built the debug version of intel-media-20.3.0 and got the following crash:
<pre>
#0  0x000000000000000e in ?? ()
#1  0x00007fffb57641a9 in CodechalDecodeAvc::DecodePrimitiveLevel (this=0x5d5a8000) at /home/chao/projects/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp:1873
#2  0x00007fffb575afc2 in CodechalDecode::Execute (this=0x5d5a8000, params=0x1ab38e10) at /home/chao/projects/media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1295
#3  0x00007fffb5a978d5 in DdiMediaDecode::EndPicture (this=0x7d225e00, ctx=0x1be35a40, context=268435456) at /home/chao/projects/media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:880
#4  0x00007fffb5a99a29 in DdiDecode_EndPicture (ctx=ctx@entry=0x1be35a40, context=context@entry=268435456) at /home/chao/projects/media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:154
#5  0x00007fffb5ac3386 in DdiMedia_EndPicture (ctx=0x1be35a40, context=268435456) at /home/chao/projects/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3828
#6  0x00007ffff60c8190 in vaEndPicture () from bin/libffmpeg-cam4linux.so
</pre>
I am confused by it though. It crashed at [codechal_decode_avc.cpp:1873](https://github.com/intel/media-driver/blob/intel-media-20.3.0/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp#L1873).
In gdb, I got
<pre>
(gdb) p m_osInterface->pfnReturnCommandBuffer
$7 = (void (*)(PMOS_INTERFACE, PMOS_COMMAND_BUFFER, uint32_t)) 0x7fffb5aef080 <Mos_Specific_ReturnCommandBuffer(_MOS_INTERFACE*, _MOS_COMMAND_BUFFER*, unsigned int)>
</pre>
How could calling Mos_Specific_ReturnCommandBuffer ends up with "0x000000000000000e"? Did I miss sth. here?

chao-camect, 2021-04-15:
I got similar crashes on a i7-1165G7, with 21.1.3 on kernel 5.11.10
<pre>
#0  0x00007f64a747af8f in mos_add_reloc_objects (reloc_target=...) at media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:596
#1  0x00007f64a7481dd1 in mos_gem_bo_process_reloc2 (bo=0x1e8fdc20) at media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:2543
#2  do_exec2 (bo=bo@entry=0x1e8fdc20, used=32768, ctx=0x59e99c80, cliprects=cliprects@entry=0x0, num_cliprects=num_cliprects@entry=0, DR4=DR4@entry=0,
    flags=0, fence=0x0) at media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:2727
#3  0x00007f64a748269a in mos_gem_bo_context_exec2 (bo=bo@entry=0x1e8fdc20, used=<optimized out>, ctx=<optimized out>, cliprects=cliprects@entry=0x0,
    at media-driver/media_driver/linux/common/os/i915/mos_bufmgr.c:2846
#4  0x00007f64a74e821c in GpuContextSpecificNext::SubmitCommandBuffer (this=0x55a54000, streamState=<optimized out>, cmdBuffer=<optimized out>,
    nullRendering=<optimized out>) at media-driver/media_driver/media_driver_next/linux/common/os/mos_gpucontext_specific_next.cpp:1290
#5  0x00007f64a748a030 in Mos_Specific_SubmitCommandBuffer (pOsInterface=0x9e8a4000, pCmdBuffer=0x7f64a6d63940, bNullRendering=0)
    at media-driver/media_driver/linux/common/os/mos_os_specific.c:4212
#6  0x00007f64a7381272 in CodechalDecodeAvcG12::DecodePrimitiveLevel (this=0x73262000)
    at media-driver/media_driver/agnostic/gen12/codec/hal/codechal_decode_avc_g12.cpp:436
#7  0x00007f64a70ed0c3 in CodechalDecode::Execute (this=0x73262000, params=0x91448010)
    at media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1341
#8  0x00007f64a74308cd in DdiMediaDecode::EndPicture (this=0x1a92f200, ctx=0x1bb94d00, context=268435456)
    at media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:913
#9  0x00007f64a7432a9f in DdiDecode_EndPicture (ctx=ctx@entry=0x1bb94d00, context=context@entry=268435456)
    at media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:173
#10 0x00007f64a745f376 in DdiMedia_EndPicture (ctx=0x1bb94d00, context=268435456)
    at media-driver/media_driver/linux/common/ddi/media_libva.cpp:3824

</pre>

YijinLiu, 2021-06-20:
Did more investigation recently. Here are what I have found:

- The crash only happens if the decoded frames are freed in another thread with certain delay (like 1s).
- It only happems on broken stream. Cannot repro on saved footage, even if I drop the packets randomly.
Here is a [minimal program](https://drive.google.com/file/d/1qKYzycBHKnNpK5U0CQ_1oIRuYXaNSBUX/view?usp=sharing) to repro it.

Try sth. like 
<pre>
./test_ihd_crash_cc "${rtsp_url}?1" "${rtsp_url}?2"  "${rtsp_url}?3"  "${rtsp_url}?4"  "${rtsp_url}?5"
</pre>
It decodes the same RTSP URL in 5 different threads so it can easily got some broken data from the RTSP url.

To make sure it's related to iHD, I run the same program:
- On i7-1165G7, it crashes.
- On i7-1165G7 with CLOSE_AVFRAME_DELAY=0 so the program frees the frames in the decoding thread. It won't crash.
- On i7-1165G7 with DISABLE_VAAPI=1 so it use software to decode the streams. It won't crash.
- On i7-1165G7 with a saved footage, it won't crash.
- On J3455, it never crashes.

Jexu, 2021-06-21:
@YijinLiu unfortunately i met some compiling issue with your minimal program, can you just share the binary to me?

YijinLiu, 2021-06-21:
You can find it [here](https://drive.google.com/file/d/1EGr9yCo6l0a8aqGQUb4qkWsG6yTEFrLs/view?usp=sharing).
I am curious what's the compiling issue though?

chao-camect, 2021-09-29:
I spent some time on this recently. Apparently, there are at least two bugs here. My previous minimal program can only repro one of them. The first one is an easy fix. It's caused by the code reads a vector w/o lock. The fix is here:
<pre>
diff --git a/media_driver/agnostic/common/os/mos_gpucontextmgr.cpp b/media_driver/agnostic/common/os/mos_gpucontextmgr.cpp
index f5ffc26..72711e3 100644
--- a/media_driver/agnostic/common/os/mos_gpucontextmgr.cpp
+++ b/media_driver/agnostic/common/os/mos_gpucontextmgr.cpp
@@ -36,10 +36,6 @@ GpuContextMgr::GpuContextMgr(GT_SYSTEM_INFO *gtSystemInfo, OsContext *osContext)
     m_gpuContextArrayMutex = MOS_CreateMutex();
     MOS_OS_CHK_NULL_NO_STATUS_RETURN(m_gpuContextArrayMutex);
 
-    MOS_LockMutex(m_gpuContextArrayMutex);
-    m_gpuContextArray.clear();
-    MOS_UnlockMutex(m_gpuContextArrayMutex);
-
     if (gtSystemInfo)
     {
         MOS_SecureMemcpy(&m_gtSystemInfo, sizeof(GT_SYSTEM_INFO), gtSystemInfo, sizeof(GT_SYSTEM_INFO));
@@ -94,11 +90,6 @@ void GpuContextMgr::CleanUp()
     if (m_initialized)
     {
         DestroyAllGpuContexts();
-
-        MOS_LockMutex(m_gpuContextArrayMutex);
-        m_gpuContextArray.clear();
-        MOS_UnlockMutex(m_gpuContextArrayMutex);
-
         m_initialized = false;
     }
 
@@ -198,15 +189,18 @@ GpuContext *GpuContextMgr::GetGpuContext(GPU_CONTEXT_HANDLE gpuContextHandle)
         return nullptr;
     }
 
+    GpuContext *gpuContext = nullptr;
+    MOS_LockMutex(m_gpuContextArrayMutex);
     if (!m_gpuContextArray.empty() && gpuContextHandle < m_gpuContextArray.size())
     {
-        return m_gpuContextArray.at(gpuContextHandle);
+        gpuContext = m_gpuContextArray.at(gpuContextHandle);
     }
     else
     {
         MOS_OS_ASSERTMESSAGE("GPU context array is empty or got invalid index, something must be wrong!");
-        return nullptr;
     }   
+    MOS_UnlockMutex(m_gpuContextArrayMutex);
+    return gpuContext;
 }
 
 void GpuContextMgr::DestroyGpuContext(GpuContext *gpuContext)
@@ -254,7 +248,9 @@ void GpuContextMgr::DestroyAllGpuContexts()
     // delete each instance in m_gpuContextArray
     for (auto &curGpuContext : m_gpuContextArray)
     {   
-        MOS_Delete(curGpuContext);
+        if (curGpuContext) {
+            MOS_Delete(curGpuContext);
+        }
     }   
 
     m_gpuContextArray.clear();  // clear whole array
diff --git a/media_driver/media_driver_next/agnostic/common/os/mos_gpucontextmgr_next.cpp b/media_driver/media_driver_next/agnostic/common/os/mos_gpucontextmgr_next.cpp
index 8d260d0..a3536b5 100644
--- a/media_driver/media_driver_next/agnostic/common/os/mos_gpucontextmgr_next.cpp
+++ b/media_driver/media_driver_next/agnostic/common/os/mos_gpucontextmgr_next.cpp
@@ -102,11 +102,6 @@ void GpuContextMgrNext::CleanUp()
     if (m_initialized)
     {   
         DestroyAllGpuContexts();
-
[... 81 lines cut ...]
However, it either crashes with the following stack trace or OOM:
<pre>
#0  0x00007ffff25405b2 in GmmLib::GmmResourceInfoCommon::GetQPitch (this=0x7fffec217240) at /tmp/4shome_tmp/gmmlib/Source/GmmLib/Resource/GmmResourceInfoCommon.cpp:1216
#1  0x00007ffff3634673 in MosInterface::GetResourceInfo (streamState=0x7fffec0f4430, resource=0x7ffff24e83e0, details=...) at /home/chao/projects/media-driver/media_driver/linux/common/os/mos_interface.cpp:1715
#2  0x00007ffff3611f40 in Mos_Specific_GetResourceInfo (pOsInterface=0x7fffec18c310, pOsResource=0x7ffff24e83e0, pResDetails=0x7ffff24e8180) at /home/chao/projects/media-driver/media_driver/linux/common/os/mos_os_specific.c:2668
#3  0x00007ffff2bd12b4 in CodecHalGetResourceInfo (osInterface=0x7fffec18c310, surface=0x7ffff24e83e0) at /home/chao/projects/media-driver/media_driver/agnostic/common/codec/hal/codechal_utilities.cpp:516
#4  0x00007ffff2c160bd in CodechalDecodeAvc::InitPicMhwParams (this=0x7fffec0deb00, picMhwParams=0x7ffff24e8ad0) at /home/chao/projects/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp:1440
#5  0x00007ffff3354d1a in CodechalDecodeAvcG12::DecodeStateLevel (this=0x7fffec0deb00) at /home/chao/projects/media-driver/media_driver/agnostic/gen12/codec/hal/codechal_decode_avc_g12.cpp:296
#6  0x00007ffff2becb17 in CodechalDecode::Execute (this=0x7fffec0deb00, params=0x7fffec0f2630) at /home/chao/projects/media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1373
#7  0x00007ffff353b3cc in DdiMediaDecode::EndPicture (this=0x7fffec03c440, ctx=0x7fffec0ef730, context=268435456) at /home/chao/projects/media-driver/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:913
#8  0x00007ffff35443c9 in DdiDecode_EndPicture (ctx=0x7fffec0ef730, context=268435456) at /home/chao/projects/media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:173
#9  0x00007ffff35c14ad in DdiMedia_EndPicture (ctx=0x7fffec0ef730, context=268435456) at /home/chao/projects/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3765
#10 0x00007ffff35ef224 in MediaLibvaInterface::EndPicture (ctx=0x7fffec0ef730, context=268435456) at /home/chao/projects/media-driver/media_driver/linux/common/ddi/media_libva_interface.cpp:328
#11 0x0000000000e28960 in vaEndPicture (dpy=0x7fffec0f7770, context=268435456) at va.c:1641
#12 0x000000000099a2aa in ff_vaapi_decode_issue (avctx=0x7fffec022640, pic=0x7fffecb6e440) at libavcodec/vaapi_decode.c:188
#13 0x0000000000650361 in vaapi_h264_end_frame (avctx=<optimized out>) at libavcodec/vaapi_h264.c:317
[... cut ...]
#16 0x000000000053f21e in decode_simple_internal (frame=0x7fffec087e80, avctx=0x7fffec022640) at libavcodec/decode.c:342
#17 decode_simple_receive_frame (frame=<optimized out>, avctx=<optimized out>) at libavcodec/decode.c:533
#18 decode_receive_frame_internal (avctx=avctx@entry=0x7fffec022640, frame=0x7fffec087e80) at libavcodec/decode.c:551
#19 0x000000000053fd30 in avcodec_send_packet (avctx=0x7fffec022640, avpkt=0x7fffec188d00) at libavcodec/decode.c:609
</pre>

I feel the right fix is to call ResetGpuContextStatus. Not sure where though. Can you guys take a look and give a suggestion?

Jexu, 2021-09-30:
My understanding, mos_bo_unreference will totally release the bo handle from kernel and make the bo unaccessible. So you maynot try to add mos_bo_unreference in GpuContextSpecificNext::ResetGpuContextStatus.  Your above log shows the resource becomes unreadable which maybe caused by the unreference call.
You can do the experiment by calling GpuContextSpecificNext::ResetGpuContextStatus in every beginning of the decodeing, such as adding it in CodechalDecode::Execute, to check if it is related to invalid bo in GpuContextSpecificNext::m_attachedResources.

chao-camect, 2021-10-07:
Both mos_bo_reference and mos_bo_unreference are reference counted. The change I proposed was to avoid a typical dangling pointer bug.
Anyway, I figured out why the fix didn't work. It's indeed a dangling pointer bug, just wrong place.
<pre>
diff --git a/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp b/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp
index ac85098..ff67c8f 100644
--- a/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp
+++ b/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp
@@ -511,6 +511,10 @@ MOS_STATUS CodechalDecodeAvc::SetPictureStructs()
     m_statusReportFeedbackNumber = picParams->StatusReportFeedbackNumber;
     m_currPic                                 = currPic;
     m_avcRefList[currPic.FrameIdx]->RefPic    = m_currPic;
+    if (m_avcRefList[currPic.FrameIdx]->resRefPic.bo != nullptr) {
+        mos_bo_unreference(m_avcRefList[currPic.FrameIdx]->resRefPic.bo);
+    }
+    mos_bo_reference(m_destSurface.OsResource.bo);
     m_avcRefList[currPic.FrameIdx]->resRefPic = m_destSurface.OsResource;
 
     // Override reference list with ref surface passed from DDI
@@ -520,6 +524,10 @@ MOS_STATUS CodechalDecodeAvc::SetPictureStructs()
     {
         if (!Mos_ResourceIsNull(&m_refFrameSurface[surfIndex].OsResource))
         {
+            if (m_avcRefList[surfIndex]->resRefPic.bo != nullptr) {
+                mos_bo_unreference(m_avcRefList[surfIndex]->resRefPic.bo);
+            }
+            mos_bo_reference(m_refFrameSurface[surfIndex].OsResource.bo);
             m_avcRefList[surfIndex]->resRefPic = m_refFrameSurface[surfIndex].OsResource;
             surfCount++;
         }
@@ -1109,6 +1117,11 @@ CodechalDecodeAvc::~CodechalDecodeAvc()
 {
     CODECHAL_DECODE_FUNCTION_ENTER;
 
+    for (int i = 0; i < CODEC_AVC_NUM_UNCOMPRESSED_SURFACE; i++) {
+        if (m_avcRefList[i] != nullptr && m_avcRefList[i]->resRefPic.bo != nullptr) {
+            mos_bo_unreference(m_avcRefList[i]->resRefPic.bo);
+        }
+    }
     CodecHalFreeDataList(m_avcRefList, CODEC_AVC_NUM_UNCOMPRESSED_SURFACE);
 
     m_osInterface->pfnDestroySyncResource(
@@ -1932,6 +1945,10 @@ MOS_STATUS CodechalDecodeAvc::DecodePrimitiveLevel()
     }
 
     // Needs to be re-set for Linux buffer re-use scenarios
+    if (m_avcRefList[m_avcPicParams->CurrPic.FrameIdx]->resRefPic.bo != nullptr) {
+        mos_bo_unreference(m_avcRefList[m_avcPicParams->CurrPic.FrameIdx]->resRefPic.bo);
+    }
+    mos_bo_reference(m_destSurface.OsResource.bo);
     m_avcRefList[m_avcPicParams->CurrPic.FrameIdx]->resRefPic =
         m_destSurface.OsResource;
 
diff --git a/media_driver/agnostic/common/codec/hal/codechal_decode_hevc.cpp b/media_driver/agnostic/common/codec/hal/codechal_decode_hevc.cpp
index b338cfa..79e12cf 100644
--- a/media_driver/agnostic/common/codec/hal/codechal_decode_hevc.cpp
+++ b/media_driver/agnostic/common/codec/hal/codechal_decode_hevc.cpp
@@ -575,6 +575,11 @@ CodechalDecodeHevc::~CodechalDecodeHevc ()
 
     m_osInterface->pfnDestroySyncResource(m_osInterface, &m_resSyncObjectWaContextInUse);
 
+    for (int i = 0; i < CODECHAL_NUM_UNCOMPRESSED_SURFACE_HEVC; i++) {
+        if (m_hevcRefList[i] != nullptr && m_hevcRefList[i]->resRefPic.bo != nullptr) {
+            mos_bo_unreference(m_hevcRefList[i]->resRefPic.bo);
+        }
+    }
     CodecHalFreeDataList(m_hevcRefList, CODECHAL_NUM_UNCOMPRESSED_SURFACE_HEVC);
 
     if (!Mos_ResourceIsNull(&m_resMfdDeblockingFilterRowStoreScratchBuffer))
@@ -1147,6 +1152,9 @@ MOS_STATUS CodechalDecodeHevc::SetFrameStates ()
 
[... 27 lines cut ...]
                     CODECHAL_DECODE_ASSERTMESSAGE("Ref frame for Current Frame is not exist. Current frame will be skipped. Thus, clear current frame Ref List.");
                     return MOS_STATUS_INVALID_PARAMETER;
@@ -2510,13 +2525,18 @@ MOS_STATUS CodechalDecodeHevc::DecodePrimitiveLevel()
     }
 
     // Needs to be re-set for Linux buffer re-use scenarios
+    if (m_hevcRefList[m_hevcPicParams->CurrPic.FrameIdx]->resRefPic.bo != nullptr) {
+        mos_bo_unreference(m_hevcRefList[m_hevcPicParams->CurrPic.FrameIdx]->resRefPic.bo);
+    }
     if (m_is8BitFrameIn10BitHevc)
     {
+        mos_bo_reference(m_internalNv12RtSurfaces[m_internalNv12RtIndexMap[m_hevcPicParams->CurrPic.FrameIdx]].OsResource.bo);
         m_hevcRefList[m_hevcPicParams->CurrPic.FrameIdx]->resRefPic =
             m_internalNv12RtSurfaces[m_internalNv12RtIndexMap[m_hevcPicParams->CurrPic.FrameIdx]].OsResource;
     }
     else
     {
+        mos_bo_reference(m_destSurface.OsResource.bo);
         m_hevcRefList[m_hevcPicParams->CurrPic.FrameIdx]->resRefPic =
             m_destSurface.OsResource;
     }
</pre>

With the two fixes patched (to be clear, the other one is calling m_gpuContextArray.at with locking), it seems to be running fine.
Will you guys take a look and merge them into the code base?

Jexu, 2021-10-08:
Can you send out both PR directly, so we can ask to verify and merge it.

chao-camect, 2021-11-02:
I think it's a fix now. Thanks!

chao-camect, 2021-11-28:
Unfortunately, commit "Correct the resource reference count" causes another problem: huge memory usage increase.
I have 12 cameras (4 8MP, 4 4MP, 4 2MP). Without the commit, gem objects use about 2G memory. With the commit, gem objects use about 6.5G memory. Is there anyway to avoid reference these bo unnecessarily?

chao-camect, 2021-12-06:
I am testing some changes like the following:
<pre>
diff --git a/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp b/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp
index ac85098..3f590fa 100644
--- a/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp
+++ b/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp
@@ -518,8 +520,11 @@ MOS_STATUS CodechalDecodeAvc::SetPictureStructs()
     uint8_t surfIndex = 0;
     while (surfCount < m_refSurfaceNum && surfIndex < CODEC_AVC_NUM_UNCOMPRESSED_SURFACE)
     {
+        m_osInterface->pfnUnReferenceResource(&m_avcRefList[surfIndex]->resRefPic);
+        m_avcRefList[surfIndex]->resRefPic = m_resInvalidRefBuffer;
         if (!Mos_ResourceIsNull(&m_refFrameSurface[surfIndex].OsResource))
         {
+            m_osInterface->pfnReferenceResource(&m_refFrameSurface[surfIndex].OsResource);
             m_avcRefList[surfIndex]->resRefPic = m_refFrameSurface[surfIndex].OsResource;
             surfCount++;
         }
</pre>
It works well for me so far, using about half of memory compared to before.
Do you guys think there are more places we can do such optimization?

Jexu, 2021-12-07:
> I am testing some changes like the following:
> 
> diff --git a/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp b/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp
> index ac85098..3f590fa 100644
> --- a/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp
> +++ b/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp
> @@ -518,8 +520,11 @@ MOS_STATUS CodechalDecodeAvc::SetPictureStructs()
>      uint8_t surfIndex = 0;
>      while (surfCount < m_refSurfaceNum && surfIndex < CODEC_AVC_NUM_UNCOMPRESSED_SURFACE)
>      {
> +        m_osInterface->pfnUnReferenceResource(&m_avcRefList[surfIndex]->resRefPic);
> +        m_avcRefList[surfIndex]->resRefPic = m_resInvalidRefBuffer;
>          if (!Mos_ResourceIsNull(&m_refFrameSurface[surfIndex].OsResource))
>          {
> +            m_osInterface->pfnReferenceResource(&m_refFrameSurface[surfIndex].OsResource);
>              m_avcRefList[surfIndex]->resRefPic = m_refFrameSurface[surfIndex].OsResource;
>              surfCount++;
>          }
> It works well for me so far, using about half of memory compared to before. Do you guys think there are more places we can do such optimization?

The issue seems related to the reference frame management. When umd try to update the reference list, it doesn't unreference the previous resource in the list. So you can check if any other places have similar reference updating. maybe in codechal_decode_avc_g12.cpp as well.

commit 5051726b (2022-02-08): [MOS] Add the lock for context array to keep thread safety

Multiple threads may access to same context array

fixes #1080
closed 2022-02-08 by intel-mediadev