intel/media-driver #1081: Support load of media-driver on unsupported platforms

dvrogozh, 2020-11-09:
From https://github.com/intel/libva/issues/466

Forwarding a [bug report](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=972833) from Debian: libva tries to load media-driver on Penryn. Since that platform is not supported by media-driver, it should not be loaded in this case.

What we need to do:
1. Remove cpu-specific flags from gmmlib build
2. Remove cpu-specific flags from driver build

dvrogozh, 2020-11-09:
Places in driver which uses specific cpu instructions:
* https://github.com/intel/media-driver/blob/f85a3d908f57c3bcbe163867cefb6bdb9959377e/media_driver/linux/common/cm/hal/cm_mem_os.h#L152
* https://github.com/intel/media-driver/blob/f85a3d908f57c3bcbe163867cefb6bdb9959377e/media_driver/agnostic/common/cm/cm_mem.h#L565
* https://github.com/intel/media-driver/blob/f85a3d908f57c3bcbe163867cefb6bdb9959377e/media_driver/cmake/linux/media_compile_flags_linux.cmake#L43

eero-t, 2020-12-08:
Is there some plan on doing run-time CPU feature detection & conditional SSE/AVX usage, for things where performance matters (such as CPU side format conversions)?

paulmenzel, 2020-12-20:
This is also Debian bug report [974772](https://bugs.debian.org/974772).

I am seeing this on a GM45 device with Pentium Dual-Core CPU T4500.

    $ lspci -nn
    00:00.0 Host bridge [0600]: Intel Corporation Mobile 4 Series Chipset Memory Controller Hub [8086:2a40] (rev 07)
    00:02.0 VGA compatible controller [0300]: Intel Corporation Mobile 4 Series Chipset Integrated Graphics Controller [8086:2a42] (rev 07)
    00:02.1 Display controller [0380]: Intel Corporation Mobile 4 Series Chipset Integrated Graphics Controller [8086:2a43] (rev 07)
    […]

Any idea, what commit introduced the regression? VLC didn’t crash in the past to my knowledge.

sebastinas, 2020-12-20:
@paulmenzel I think you need to install `i965-va-driver` and remove `intel-media-va-driver`.

paulmenzel, 2020-12-26:
> @paulmenzel I think you need to install `i965-va-driver` and remove `intel-media-va-driver`.

Yes, thank you. *i965-va-driver-shaders* is already installed, and MPEG2 decoding is available. (H264 support was never finished.)

dvrogozh, 2021-01-27:
While installing `i965-va-driver` might provide a workaround, our side desire is to get this behavior fixed in `intel-media-va-driver` since that's the driver we have on a new platforms.

As a step to the ultimate fix we've merged https://github.com/intel/media-driver/commit/ff55f781c91509ad6b6d21cd357346656066740b. This change removes cpu instruction set specific compile options from driver build. Those few places which really require specific instruction sets are compiled separately and we added code to dynamically check whether underlying instruction set is actually supported by the CPU. This fix is available starting from `>= intel-media-21.1.0` release. I recommend to update.

This being said, I suspect that **issue might not be fully solved yet** for the reason that `gmmlib` from which driver depends on was not fixed in the same way. Hence, if there is any call to gmmlib (and it has similar issue) we will see similar SIGILL failure, but in different place. We would appreciate a help if someone will try this out and report the actual behavior back to us.

paulmenzel, 2021-01-27:
Thank you for the update. You might be able to reproduce this issue in QEMU.

egorovdanil, 2021-03-11:
I tried to run `vainfo` on an unsupported platform (amd) and found out that the exit from `intel-media-va-driver` occurs before any entry into `gmmlib`:
`DdiMedia__Initialize()` exit location: [media_libva.cpp#L1736](https://github.com/intel/media-driver/blob/f60361dad30eb1c58aabaab65da514bd2b27fbaf/media_driver/linux/common/ddi/media_libva.cpp#L1736)
`gmmlib` entry point location in `DdiMedia__Initialize()`: [media_libva.cpp#L1823](https://github.com/intel/media-driver/blob/98caf5cee907fc5246c611833958ad316d4d64db/media_driver/linux/common/ddi/media_libva.cpp#L1823)

XinfengZhang, 2023-05-08:
https://github.com/intel/media-driver/pull/1446 as first step

sebastinas, 2023-05-08:
The approach in #1446 does not help us (Debian). Intrinsics can be used and that's not the problem. The build just needs to ensure that `vaDriverInit` is able to detect whether the GPU is supported without using those features. After that it's perfectly fine to use them.

dvrogozh, 2023-05-09:
@XinfengZhang : when we last looked into this story 2 years ago, we left in a state that changes to media-driver were done, but follow up was needed for gmmlib. Maybe something has changed during 2 years, but I still think that the pattern we followed should be continued: we need runtime checks for supported features for both media driver and gmmlib.

XinfengZhang, 2023-05-10:
#1446  is using to enable media driver on non-x86 CPU platform, it is for D-graphics which is plugged in non-x86 platform. but it could be used to remove the dependencies of x86 intrinsics, gmmlib also have such flags to remove the dependencies. 

@sebastinas  I suppose you want a auto CPU (not GPU)feature detection in runtime? if it is GPU feature, vaDriverInit will failed on un-support GPU. 

so, a quick summary, you want a runtime check of CPU feature to enable/disable some media driver feature which depend on these CPU feature. :)

sebastinas, 2023-05-10:
No. We need that vaDriverInit is able to run without any additional CPU features. If the GPU is supported, the CPU features are automatically supported. If the GPU is not supported, vaDriverInit need bails out. But all the checks if the GPU is supported or not, may not use any additional CPU features beyond the baseline.

sebastinas, 2023-05-11:
@eero-t Okay, this may have changed. Last time I looked at it (2-3 years ago), that would have been enough.