intel/media-driver #1083: [Gen12 regression][sample_vpp]The md5 of input file and output file are same through sample_vpp sharpen for I420 and YV12 on Gen12.

yefengxx, 2020-11-16:
The md5 of input file and output file are same through sample_vpp sharpen for **I420** and **YV12** on **Gen12**.   
for example:
sample_vpp -sw 3840 -sh 2160 -scc yv12 -dw 3840 -dh 2160 -dcc yv12 -detail 75 -i input -o output
On ICL platform, we can get different input and output files for **I420** and **YV12**, and we confirm that this is the correct result. But on **Gen12** the md5 of input file and output file are same.

Sherry-Lin, 2020-11-17:
You mentioned it's regression. Could you provide the good stack commit list? Thanks.

yefengxx, 2020-11-17:
> You mentioned it's regression. Could you provide the good stack commit list? Thanks.

I'm sorry, maybe I didn't express it clearly enough. I want to say that there is no issue with the ICL platform, but GEN12(TGL&DG1) found this issue.

XinfengZhang, 2020-11-18:
@FurongZhang , please help to check the issue

MicroYY, 2021-11-09:
Hi @yefengxx 
This is expected, as VPP BLT doesn't gives bit match result even in/out is the same. 
This case is using render kernel. For ICL and TGL, different kernels are used. 
So there may be results that ICL in/out bit match while TGL not.

closed 2021-11-23 by MicroYY