intel/media-driver #1085: Report caps in vainfo if no Guc binaries in machine

Sherry-Lin, 2020-11-16:
There is no Guc binaries in some users' environment but some media encode features are rely on Huc. Is it possible to report the caps in vainfo if no Guc in local?

XinfengZhang, 2020-11-18:
I understand the request. sometimes , driver need to query some information like huc/guc readiness.  but how about the usage case? application query the readiness of huc/guc readiness , then will have different behavior?

XinfengZhang, 2020-11-24:
sorry, understanding is incorrect. if there are no huc, driver should change the capability report , such as the BRC mode ...

Jexu, 2022-12-02:
This may depend on i915 exposing api to query guc/ huc loading state.

XinfengZhang, 2024-05-20:
we could use I915_PARAM_HUC_STATUS to apply it, but who are caring this feature?