intel/media-driver #1087: [KBL] tonemap_vaapi filter not working as expected

nyanmisaka, 2020-11-18:
As I see that `tonemap_vaapi` filter have been implemented in ffmpeg 4.3.
Then I tried it to do H2S for my HDR videos on a KabyLake chip(HD 630), ffmpeg works with no error but the output color doesn’t seem to have been changed at all.

The 4k HDR example is from https://4kmedia.org/sony-food-fizzle-uhd-hdr-4k-demo/
Note that the software tonemap filter and tonemap_opencl filter are all works for this video.
I am using libva 2.8 and media-driver 20.2 comes with Ubuntu 20.10.

HDR has been enabled in iHD driver for KBL.https://github.com/intel/media-driver/commit/c82f434c7ba97de08174736a654e5acfe8d67099
Does KBL support HDR Tonemapping? Or it is only supported on ICL+? Thanks in advance!

```
ffmpeg -hwaccel vaapi -hwaccel_output_format vaapi \
-vaapi_device /dev/dri/renderD128 -i HDR.mp4 -c:v h264_vaapi \
-b:v 8000000 -level 41 -profile:v high -an -sn \
-vf 'scale_vaapi=w=1920:h=1080,tonemap_vaapi=t=bt709:format=nv12' \
-y /tmp/SDR.mp4
```

nyanmisaka, 2020-11-18:
```
    for (i = 0; i < num_query_caps; i++) {
        if (hdr_cap[i].metadata_type != VAProcHighDynamicRangeMetadataNone)
            break;
    }
```

https://github.com/FFmpeg/FFmpeg/blob/e3081d6f4f36adc3d06b88a3c43ca6f18ab674c8/libavfilter/vf_tonemap_vaapi.c#L204

@FurongZhang 
The iHD driver report that it supports HDR tonemapping on KBL however the color of output video doesn't change at all.

Could you help me to identify this on KBL?

XinfengZhang, 2020-11-18:
@FurongZhang , could you help to check it?

FurongZhang, 2021-01-07:
Sure. I will take a look at this.

uartie, 2021-01-07:
H2S on KBL works for me with gstreamer-vaapi:

```
gst-launch-1.0 -vf filesrc location=./Sony\ Food\ Fizzle\ UHD\ 4K\ Demo.mp4 \
 ! qtdemux ! h265parse ! vaapih265dec ! vaapipostproc ! vaapisink
```

uartie, 2021-01-07:
> HDR H2S tone mapping on KBL works for me with gstreamer-vaapi:
> 
> ```
> gst-launch-1.0 -vf filesrc location=./Sony\ Food\ Fizzle\ UHD\ 4K\ Demo.mp4 \
>  ! qtdemux ! h265parse ! vaapih265dec ! vaapipostproc ! vaapisink
> ```

My KBL is i7-8650U (UHD 620).  Can you verify if it works for you on gstreamer-vaapi?  Maybe this is an issue in ffmpeg?

uartie, 2021-01-07:
This also works for me on KBL... gstreamer-vaapi tone map vs. _**no**_ tone map side-by-side (note: requires scaling to 1080p due to max resolution limitation in vaapioverlay):
```
gst-launch-1.0 -vf filesrc location=./Sony\ Food\ Fizzle\ UHD\ 4K\ Demo.mp4 \
  ! qtdemux ! h265parse ! vaapih265dec ! tee name=decoded ! queue \
  ! vaapipostproc hdr-tone-map=auto \
  ! "video/x-raw(memory:VASurface),width=1920,height=1080" \
  ! vaapioverlay name=overlay sink_1::xpos=1920 \
  ! vaapisink fullscreen=1 decoded. ! queue \
  ! vaapipostproc hdr-tone-map=disabled \
  ! "video/x-raw(memory:VASurface),width=1920,height=1080" \
  ! overlay.
```

uartie, 2021-01-07:
> This also works for me on KBL... gstreamer-vaapi tone map vs. _**no**_ tone map side-by-side (note: requires scaling to 1080p due to max resolution limitation in vaapioverlay):
>
> ```
> gst-launch-1.0 [...]

[image]

uartie, 2021-01-07:
@nyanmisaka do you have the driver with the patch from #821 (i.e. 73bcc3415e32fa0697e384aa1f301a8b56327bc1)?

nyanmisaka, 2021-01-07:
> @nyanmisaka do you have the driver with the patch from #821 (i.e. [73bcc34](https://github.com/intel/media-driver/commit/73bcc3415e32fa0697e384aa1f301a8b56327bc1))?

Sure. I'm using media-driver 20.2 and this commit was introduced in `intel-media-20.1.pre2`.

nyanmisaka, 2021-01-07:
> This also works for me on KBL... gstreamer-vaapi tone map vs. _**no**_ tone map side-by-side (note: requires scaling to 1080p due to max resolution limitation in vaapioverlay):
>
> ```
> gst-launch-1.0 [...]

Unfortunately the color is still washed out in gstreamer.
Does it mean UHD 6XX can do tonemapping but HD 6XX cannot?
Or is there anything I should update?
```
gstreamer-vaapi 1.18.0
libva 2.8
iHD 20.2
```

[image]

```
gst-inspect-1.0 vaapi
Plugin Details:
  Name                     vaapi
  Description              VA-API based elements
  Filename                 /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstvaapi.so
  Version                  1.18.0
  License                  LGPL
  Source module            gstreamer-vaapi
  Source release date      2020-09-08
  Binary package           gstreamer-vaapi
  Origin URL               https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/issues

  vaapijpegdec: VA-API JPEG decoder
  vaapimpeg2dec: VA-API MPEG2 decoder
  vaapih264dec: VA-API H264 decoder
  vaapivc1dec: VA-API VC1 decoder
  vaapivp8dec: VA-API VP8 decoder
  vaapivp9dec: VA-API VP9 decoder
  vaapih265dec: VA-API H265 decoder
  vaapioverlay: VA-API overlay
  vaapipostproc: VA-API video postprocessing
  vaapidecodebin: VA-API Decode Bin
  vaapisink: VA-API sink
  vaapimpeg2enc: VA-API MPEG-2 encoder
  vaapih264enc: VA-API H264 encoder
  vaapijpegenc: VA-API JPEG encoder
  vaapivp8enc: VA-API VP8 encoder
  vaapih265enc: VA-API H265 encoder

  16 features:
  +-- 16 elements
```

```
vainfo
libva info: VA-API version 1.8.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_8
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.8 (libva 2.8.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 20.2.0 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
```

uartie, 2021-01-07:
> Unfortunately the color is still washed out in gstreamer.
> Does it mean UHD 6XX can do tonemapping but HD 6XX cannot?
> Or is there anything I should update?

@FurongZhang it looks like @nyanmisaka software stack is correct... so maybe it's driver/platform issue?

uartie, 2021-01-07:
> > Unfortunately the color is still washed out in gstreamer.
> > Does it mean UHD 6XX can do tonemapping but HD 6XX cannot?
> > Or is there anything I should update?
> 
> @FurongZhang it looks like @nyanmisaka software stack is correct... so maybe it's driver/platform issue?

Hmmm... I just tested using gstreamer on a KBL i7-7700 with HD630 graphics and it worked for me.  But my software stack is latest master branch code:
```
libva (master) heads/master-0-g0721109 
gmmlib (master) tags/intel-gmmlib-20.4.1-0-g6f15b79 
intel-media-driver (master) heads/master-0-gfcdacb40 
msdk (master) heads/master-0-ga0709387 
gstreamer (master) heads/master-0-ge02902a78 
gst-plugins-base (master) heads/master-0-gf573d9123 
gst-plugins-good (master) heads/master-0-g79efd372c 
gst-plugins-bad (master) heads/master-0-g3e2e07bb8 
gstreamer-vaapi (master) heads/master-0-g5523b755 
libva-utils (master) heads/master-0-g83728b9
```

uartie, 2021-01-07:
...and gstreamer-vaapi tone map still works for me if I use libva 2.8, iHD 20.2, and gstreamer 1.18:
```
libva (v2.8) tags/2.8.0-0-g963a4b6 
gmmlib (master) tags/intel-gmmlib-20.4.1-0-g6f15b79 
intel-media-driver (intel-media-20.2) tags/intel-media-20.2.0-0-g97e2ea99 
gstreamer (1.18) heads/1.18-0-g16178d46c 
gst-plugins-base (1.18) heads/1.18-0-g308247297 
gst-plugins-good (1.18) heads/1.18-0-gd432ca10d 
gst-plugins-bad (1.18) heads/1.18-0-ga35760806 
gstreamer-vaapi (1.18) tags/1.18.0-0-gde6fb609 
libva-utils (master) heads/master-0-g83728b9
```

So I'm a little stumped why it doesn't work for you.

@FurongZhang sorry for hijacking this issue... maybe you can reproduce and/or resolve better than me?

nyanmisaka, 2021-01-25:
Hey! The issue is caused by some unmerged patches from https://github.com/intel-media-ci/intel-ffmpeg-patch

`tonemap_vaapi` in ffmpeg works well now with them applied.

I have no idea with the issue in gstreamer, but the original issue has been resolved. Close it for now.

closed 2021-01-25 by nyanmisaka