intel/media-driver #1089: Build failure

dbermond, 2020-11-19:
I'm getting the following build error on the git master branch (currently at commit c5bd97d4):

```
[100%] Built target iHD_drv_video_OBJ
Scanning dependencies of target iHD_drv_video_STATIC
Scanning dependencies of target iHD_drv_video
[100%] Linking CXX static library iHD_drv_video.a
[100%] Linking CXX shared library iHD_drv_video.so
[100%] Built target iHD_drv_video_STATIC
[100%] Built target iHD_drv_video
Scanning dependencies of target RunULT
Running devult...
./devult: symbol lookup error: ../libdrm_mock/libdrm_mock.so: undefined symbol: mos_gem_bo_is_softpin
make[2]: *** [media_driver/linux/ult/ult_app/CMakeFiles/RunULT.dir/build.make:82: RunULT] Error 127
make[1]: *** [CMakeFiles/Makefile2:373: media_driver/linux/ult/ult_app/CMakeFiles/RunULT.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
```

A git bisect shows that this is caused by commit ed04dbcc1d45df4ae88caa8c5565fb3864c1f167.

System information:
- OS: Arch Linux x86_64
- Compiler: gcc 10.2.1
- gmmlib: git master
- libva: git master

XinfengZhang, 2020-11-23:
@Jister , please fix the ult issue , thanks

Jister, 2020-11-23:
@dbermond Could you have a quick try with #1091

dbermond, 2020-11-23:
> Could you have a quick try with #1091

I can confirm that it fixes this issue for me.

Thank you for fixing this.

PR #1091 (2020-11-27): Fix ult build issue.
Fix #1089
closed 2020-11-27 by intel-mediadev