intel/media-driver #1095: Crash when using vo=vaapi in mpv

LaserEyess, 2020-11-28:
For as far back as I have tried, `--vo=vaapi` is crashing when using `LIBVA_DRIVER_NAME=iHD`. It does not crash when using `LIBVA_DRIVER_NAME=i965`, however, it is broken in a different way. There is certainly an mpv bug or two here, but I think there may be an intel one as well. 

Information about what `--vo=vaapi` is can be found here: https://mpv.io/manual/master/#video-output-drivers-vaapi

It is mostly used as a legacy driver for support of ancient hardware but it is (barely) supported in mpv.

### Hardware/Software
 - CPU: Intel(R) Core(TM) i7-7500U CPU @ 2.70GHz
 - GPU: Intel Corporation HD Graphics 620 (rev 02)
 - Kernel: 5.9.10-arch1-1
 - Distro: Arch Linux
 - Mesa: 20.2.3
 - libva: 2.9.1
 - intel-media-driver: 20.4.1
 - mpv:  0.33.0-5-g62fb374349 (can be reproduced at least as far back as 0.30.0)

### Reproduction Steps
```
LIBVA_DRIVER_NAME=iHD mpv --no-config --hwdec=no --vo=vaapi a_video.mkv
```
Any video should reproduce this

### Backtrace
https://0x0.st/i7zE.txt

Let me know what else is useful.

XinfengZhang, 2020-11-29:
thanks for reporting, we will check it, but the backtrace is broken? there are no code from *3 to *2, I suppose the issue exist in the  DdiMedia_PutSurface instead of DdiMediaUtil_AllocateSurface . it will call get_drawable to render the picture. will try to reproduce it

LaserEyess, 2020-11-29:
You are right, that backtrace was broken. I rebuilt the same versions libva/intel-media-driver with debug symbols but I used an old coredump and that was giving an incorrect trace. I rebuilt mpv against the newly built libraries just in case to get a new backtrace: https://0x0.st/i7XZ.txt.

I can reproduce this at will, so anything else you need let me know.

wangyan42164, 2020-12-03:
@LaserEyess 
I tried your command with one HEVC 4K MKV file in my hand. But I cannot find the crash and play rightly (with vaapi-x11).

CPU: Intel(R) Core(TM) i7-7700 CPU @ 3.60GHz
GPU: Intel Corporation HD Graphics 630 (rev 04)
Kernel: 5.7.7 which is built by myself. The source code comes from kenerl.org
Distro: Fedora 28 (Workstation Edition)
Mesa: From F28 repo.
libva: latest master branch: 6fd92cfa453f408b631d68165290bd08eb68c5f9
gmmlib: latest master branch: 1598c1b46fb911cb9060d1a5380aec1ca1ef2c19
media-driver: atest master branch: 91baad7644beef6ae42b595031bd3a1a5a1aa08f

For mpv, I tried the following 2 versions:
release/0.33 branch: 0728b514980cccd13543eea53a8e23332e233a6c
master: 84d0930fa1489868ef3034b5ae18ba760b1a2db9

I am not sure if your clip has some special syntax elements?

Thanks.
Yan Wang

wangyan-intel, 2020-12-03:
mpv 0.33.0-22-g84d0930fa1 Copyright © 2000-2020 mpv/MPlayer/mplayer2 projects
 built on Thu Dec  3 16:04:42 CST 2020
FFmpeg library versions:
   libavutil       56.61.100
   libavcodec      58.114.100
   libavformat     58.64.100
   libswscale      5.8.100
   libavfilter     7.91.100
   libswresample   3.8.100
FFmpeg version: N-100131-g0639f5c294

LaserEyess, 2020-12-03:
@wangyan42164 
I'm able to reproduce this crash on any video, regardless of codec. For example, if you have youtube-dl you can recreate it with any such video, such as this randome one for example: https://www.youtube.com/watch?v=NUYvbT6vTPs

I downloaded this random video locally and the same issue persists. If you don't have youtube-dl you can download it from here https://0x0.st/i7UF.mkv. Once again the command is simply:
```
LIBVA_DRIVER_NAME=iHD mpv --no-config --vo=vaapi --hwdec=no https://0x0.st/i7UF.mkv
```

My version of gmmlib is 20.3.3, other libraries I listed above. Mpv uses libva directly for `--vo=vaapi` so I don't think that's relevant. I will attempt to rebuild all of those libraries from source and link them to mpv from source as well to see if I can re-create the crash, but that will have to wait until this weekend.

wangyan42164, 2020-12-04:
@LaserEyess 
I tried your command
The playback isn't fluent because network speed is slow but no error.
[yanwang@yanwang-kbl mpv]$ LIBVA_DRIVER_NAME=iHD mpv --no-config --vo=vaapi --hwdec=no https://0x0.st/i7UF.mkv
 (+) Video --vid=1 (*) (vp9 1920x1080 60.000fps)
 (+) Audio --aid=1 --alang=eng (*) (aac 2ch 44100Hz)
[vo/vaapi] OSD format not supported. Disabling OSD.
[vo/vaapi] Warning: this compatibility VO is low quality and may have issues with OSD, scaling, screenshots and more.
[vo/vaapi] vo=gpu is the preferred choice in any case and includes VA-API support via hwdec=vaapi or vaapi-copy.
AO: [alsa] 44100Hz stereo 2ch float
VO: [vaapi] 1920x1080 yuv420p
(Buffering) AV: 00:00:10 / 00:02:54 (6%) A-V:  0.000 Dropped: 78 Cache: 0.1s/26KB
[ffmpeg] https: Stream ends prematurely at 3161504, should be 49025833
[ffmpeg] https: Will reconnect at 3161504 in 0 second(s), error=Input/output error.

Exiting... (Quit)

LaserEyess, 2020-12-04:
I'll attempt to rebuild mpv linked against those specific library commits you posted earlier. Thanks for being so responsive, I'll try to narrow this down on my end.

LaserEyess, 2020-12-05:
- libgmm: https://github.com/intel/gmmlib/commit/8057a638c1bdf973b29c98dc9ebf5822a514e80c
 - media-driver: https://github.com/intel/media-driver/commit/b7b1b00619e135d613ad563876fbe0506db63341
 - libva: https://github.com/intel/libva/commit/179a07b6c8642a53e274210a78912a28744b424d
 - mpv: https://github.com/mpv-player/mpv/commit/7c4465cefb27d4e0d07535d368febdf77b579566

With rebuilding all of these libraries and then mpv, I still get the same crash: https://0x0.st/ihiu.txt

Some other things I thought of that might account for differences in our environment:
1. I am using sway (wayland), with Xwayland version 1.20.10
2. My default CFLAGS/CXXFLAGS might produce bad behavior (??)
```
CPPFLAGS="-D_FORTIFY_SOURCE=2"
CFLAGS="-march=native -O2 -pipe -fno-plt"
CXXFLAGS="-march=native -O2 -pipe -fno-plt"
LDFLAGS="-Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now"
```
3. Mesa has fixed [some vaapi bugs](https://gitlab.freedesktop.org/mesa/mesa/-/issues?scope=all&utf8=%E2%9C%93&state=all&label_name[]=VA-API) between now and 2018 (when fedora 28 was released).

I will attempt to rebuild with no `-march=native` to see if that makes a difference, since I think that's the most suspect part of our env differences outside of wayland.

wangyan-intel, 2020-12-07:
@LaserEyess From your CFLAGS/CXXFLAGS/LDFLAGS, I cannot find any side effect.  Could you try it based on X? I am not sure if it is wayland/Xwayland issue. IMHO, current media driver should not support Wayland when calling putSurface. @XinfengZhang will provide one unmerged patch for wayland support. You could try it too.

LaserEyess, 2020-12-07:
Ok, I tried Xorg directly instead of via Xwayland and you are right, there is no crash. Everything works perfectly, even on i965. However, Xwayland is supposed to be Xorg, it is there precisely for legacy clients to use the Xorg server on a wayland compositor. Is there enough of a difference that this crash could occur?

wangyan-intel, 2020-12-08:
@LaserEyess 
Yes. VAAPI only supports dri2 so far. I have also confirmed with @XinfengZhang. He will give more details.

LaserEyess, 2020-12-08:
Just to clarify a couple things, This actually works on radeonsi ([sort of ](https://github.com/intel/libva/issues/278#issuecomment-735481005)), so is it VAAPI/libva that does not support this or just intel-media-driver? Also, is the crash still an intel-media-driver bug or is there someone else I need to report it to?

XinfengZhang, 2020-12-12:
we have not add dri3 support with libva, also , have not enable wayland support inside media-driver. i965 enabled wayland support, but you mentioned that there are another different issue?

LaserEyess, 2020-12-13:
i965 exits with:
```
[vaapi] libva: /usr/lib/dri/i965_drv_video.so init failed
[vaapi] Failed to initialize VAAPI: unknown libva error
```
Given the discussion in this thread I would say that's actually expected, i965 does not support dri3. 

However, I will say that libva itself does seem to support wayland, at least using mesa's radeonsi vaapi driver. So I'm not quite sure how that's possible if you're saying libva does not support it *at all*. But that doesn't really matter, I still think this is a bug because media-driver or libva should exit gracefully and not crash.

XinfengZhang, 2020-12-13:
agree, media-driver or libva should exist gracefully without crash. the easiest way is to add some protection/check code.If it's convenient,  could you provide a PR for it?  because it is reproducible on your site. of course. of course , we also could try to provide a patch, but suppose need your help to verify it.

LaserEyess, 2020-12-16:
I'm afraid I'm not familiar enough with how to do this. How would this check work? Is it possible to just detect wayland and exit? That wouldn't really fix the problem I guess. If you can provide a patch I can certainly test it but I would need an incredible amount of handholding and explanation to make a PR on my own.

wangyan-intel, 2020-12-21:
@LaserEyess Could you please try @Jexu's PR? Thanks.

closed 2021-01-04 by intel-mediadev

LaserEyess, 2021-08-11:
@wangyan-intel @XinfengZhang @Jexu I apologize for not even responding to this, that was rude of me, but I was having trouble testing the patch and I gave up and forgot about this. However, 8 months later, I am sorry to say that this bug is not fixed by that patch. I want to get this fixed though because I'm ultimately interested in having good vaapi support for wayland, and I think "not having random crashes on wayland" is a good start. 

I have rebuilt a libva (https://github.com/intel/libva/commit/6e86b4fb4dafa123b1e31821f61da88f10cfbe91) and this repository (https://github.com/intel/media-driver/commit/f315c36020eed2d2b52b9557cb888ca535f17886) and got mpv to work with them via `LD_PRELOAD` and `LIBVA_DRIVERS_PATH`. Both of them have full debug symbols and I can set breakpoints if needed to step through the code.

Here is a current backtrace: https://0x0.st/-J1l.txt

Jexu's PR fixed an assert the line right before the crash, it calls into [`dri_vtable->get_drawable()`](https://github.com/intel/media-driver/blob/master/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp#L392) here and crashes.

It looks like what happens is that inside that function is that it immediately calls into a funciton called `do_drawable_hash` in libva. In there it hits [this line](https://github.com/intel/libva/blob/master/va/x11/va_dricommon.c#L74), and the problem there is
```
(gdb) info locals
dri_state = 0x7fab78023100
index = 2
dri_drawable = 0x0
(gdb) p dri_state->createDrawable
$1 = (struct dri_drawable *(*)(VADriverContextP, XID)) 0x0
```
So that's the issue I guess, a NULL pointer as this function? I did not look any further than this, but it looks like `dri_state` is [created here](https://github.com/intel/libva/blob/master/va/x11/va_dricommon.c#L64).

What's the next step here? Is there a way to compile libva with ASAN or some deeper debugging information? Should I report a new bug to libva?

Jexu, 2021-08-11:
> we have not add dri3 support with libva, also , have not enable wayland support inside media-driver. 

so you saw the NULL pointer when using wayland. @XinfengZhang  can correct me if i miss something.

LaserEyess, 2021-08-11:
Yes the scenario here is using mpv's `--vo=vaapi` on **Xwayland**, intel specifically. Here's a list of things that work and do not work.

### Works
On Xwayland (via wayland)
```
LIBVA_DRIVER_NAME=i965 mpv --no-config --vo=vaapi $file
```
(Does not play video, but exits gracefully)

On Xwayland (via wayland) on an AMD system
```
LIBVA_DRIVER_NAME=radeonsi mpv --no-config --vo=vaapi $file
````
Plays video

On Xorg
```
LIBVA_DRIVER_NAME=iHD mpv --no-config --vo=vaapi $file
```
As expected, this has no issues since it isn't Xwayland

### Crashes
On Xwayland (via wayland)
```
LIBVA_DRIVER_NAME=iHD mpv --no-config --vo=vaapi $file
```
Crashes as above

To be clear I do not expect ``--vo=vaapi`` to work on wayland, or even Xwayland, considering it's dri3. However it should not crash.

Jexu, 2021-08-12:
look like that pointer dri_state->createDrawable does not init.  You can report a bug to libva, and we may add null pointer check here before trying to call this func.

PR #1112 (2021-01-04): [Decode] Fix media driver crash by return error code instead of assert
currently media-driver doesn’t support Wayland/XWayland and Dri3. (Only for X and Dri2).

fixes #1095
commit 343ad521 (2021-01-04): [Decode] Fix media driver crash by return error code instead of assert

currently media-driver does not support Wayland/XWayland and Dri3. (Only for X and Dri2).

fixes #1095