intel/media-driver #1096: Import or use of X-tiled BGRX surfaces has regressed

fhvwy, 2020-12-01:
The hardware is an Ice Lake 1065G7.  I'm running Xorg on a 1920x1080 display, and it makes its framebuffers as X-tiled BGRX.  The VAAPI driver is iHD at versions below, gmmlib is always 1598c1b46fb911cb9060d1a5380aec1ca1ef2c19 (current head).

`# ffmpeg_g -v 55 -y -f kmsgrab -device /dev/dri/card0 -i - -vf 'hwmap=derive_device=vaapi,scale_vaapi=format=nv12' -c:v h264_vaapi out.mp4`

At 1c8e076cad591702a0faa1fae1f429e517d33811 (from a month ago, the last time I rebuilt the driver), [output is correct](http://ixia.jkqxz.net/~mrt/8dd0d1cc8d7c986651b7af7a0c460f94/output-at-1c8e076c.jpeg).

At 55268ab458e20365f61d3c636e5eb2597e2d3a30 (current head), [output is broken](http://ixia.jkqxz.net/~mrt/8dd0d1cc8d7c986651b7af7a0c460f94/output-at-55268ab4.jpeg), and looks like it has lost the tiling mode somehow.

It doesn't seem to make a difference whether the import happens with `DRM_PRIME` (`VASurfaceAttribExternalBuffers`, using an implicit modifier) or `DRM_PRIME_2` (`VADRMPRIMESurfaceDescriptor`, including `I915_FORMAT_MOD_X_TILED` explicitly).

fhvwy, 2020-12-01:
Working in 1f70472c765c086aef77d7ca9604d2754c333ec2.

Broken by fe06066f8c643b75d6cdac21df8e16106d74e89c.

Maybe the gmmlib API change there has lost the tiling information in some cases?

xry111, 2020-12-26:
#1113 fixed this issue on my laptop (with an ICL 1065G7).

XinfengZhang, 2020-12-31:
@fhvwy , it should already been fixed. could you help to try #1113

fhvwy, 2021-01-06:
Yep, the X-tiled surface import is fixed in the current version (I tested fcdacb40e128dfeaa95d6c9166d547dea72221ac with gmmlib 6f15b795e1511febf233b42d7dde603f52fcfdab).  Thank you!

Jexu, 2022-12-02:
I will close issue now since it has been resolved. Thanks for reporting and feel free to re-open it agsin if have any other questinos.

closed 2022-12-02 by Jexu