intel/media-driver #1103: media-driver-intel-media-20.4.3 build failed, has no member named 'CreateCustomResInfoObject'

heitbaum, 2020-12-08:
20.4.2 builds fine. 20.4.3 doesn’t.

http://paste.ubuntu.com/p/j527yybrF2/

This is being build in a docker container - Ubuntu 18.04.5 LTS"
Linux 10c557ef484f 5.9.12 #1 SMP Mon Dec 7 14:05:54 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

```
../media_driver/linux/common/ddi/media_libva_util.cpp:175:5: error: 'GMM_RESCREATE_CUSTOM_PARAMS' was not declared in this scope; did you mean 'GMM_RESCREATE_PARAMS'?
  175 |     GMM_RESCREATE_CUSTOM_PARAMS gmmCustomParams;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     GMM_RESCREATE_PARAMS
../media_driver/linux/common/ddi/media_libva_util.cpp:482:29: error: 'gmmCustomParams' was not declared in this scope
  482 |             MOS_ZeroMemory(&gmmCustomParams, sizeof(gmmCustomParams));
      |                             ^~~~~~~~~~~~~~~
../media_driver/linux/common/ddi/media_libva_util.cpp:528:63: error: 'GMM_CLIENT_CONTEXT' {aka 'class GmmLib::GmmClientContext'} has no member named 'CreateCustomResInfoObject'; did you mean 'CreateResInfoObject'?
  528 |             gmmResourceInfo = mediaDrvCtx->pGmmClientContext->CreateCustomResInfoObject(&gmmCustomParams);
      |                                                               ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                               CreateResInfoObject
[6/208] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/os/mos_context_specific
```

The build environment is driven by https://libreelec.wiki/development-1/build-basics and using master from https://github.com/LibreELEC/LibreELEC.tv

heitbaum, 2020-12-08:
Bumping gmmlib: update to 20.3.3 has resolved this (using libva 1.10.0 already) - will test. Can we update the dependencies?

XinfengZhang, 2020-12-31:
good suggestion, we will check with gmm team to add version and version check.  thanks

closed 2021-03-09 by XinfengZhang