intel/media-driver #1104: [Gst-vaapi] DMA->VAAPI and DMA->SysMEM conversion worked on CFL platform, but failed on TGL platform

FocusLuo, 2020-12-09:
Detail: 
On CFL
gst-launch-1.0 -vf filesrc location=/home/focus/workspace_focus/full/hevc/ra_main-BasketballDrillText_832x480_50_qp22.bin  ! h265parse ! vaapih265dec ! "video/x-raw(memory:VASurface)" ! vaapipostproc ! "video/x-raw(memory:VASurface)" ! vaapih264enc ! h264parse ! filesink location=test1.h264 (worked) 
gst-launch-1.0 -vf filesrc location=/home/focus/workspace_focus/full/hevc/ra_main-BasketballDrillText_832x480_50_qp22.bin  ! h265parse ! vaapih265dec ! "video/x-raw(memory:VASurface)" ! vaapipostproc ! "video/x-raw" ! vaapih264enc ! h264parse ! filesink location=test1.h264 (worked) 
gst-launch-1.0 -vf filesrc location=/home/focus/workspace_focus/full/hevc/ra_main-BasketballDrillText_832x480_50_qp22.bin  ! h265parse ! vaapih265dec ! "video/x-raw(memory:DMABuf)" ! vaapipostproc ! "video/x-raw(memory:VASurface)" ! vaapih264enc ! h264parse ! filesink location=test1.h264 (DMA->VAAPI worked) 
gst-launch-1.0 -vf filesrc location=/home/focus/workspace_focus/full/hevc/ra_main-BasketballDrillText_832x480_50_qp22.bin  ! h265parse ! vaapih265dec ! "video/x-raw(memory:DMABuf)" ! vaapipostproc ! "video/x-raw" ! vaapih264enc ! h264parse ! filesink location=test1.h264 (DMA->SysMEM  worked) 

On TGL:
gst-launch-1.0 -vf filesrc location=/home/focus/workspace_focus/full/hevc/ra_main-BasketballDrillText_832x480_50_qp22.bin  ! h265parse ! vaapih265dec ! "video/x-raw(memory:VASurface)" ! vaapipostproc ! "video/x-raw(memory:VASurface)" ! vaapih264enc ! h264parse ! filesink location=test1.h264 (worked) 
gst-launch-1.0 -vf filesrc location=/home/focus/workspace_focus/full/hevc/ra_main-BasketballDrillText_832x480_50_qp22.bin  ! h265parse ! vaapih265dec ! "video/x-raw(memory:VASurface)" ! vaapipostproc ! "video/x-raw" ! vaapih264enc ! h264parse ! filesink location=test1.h264 (worked) 
gst-launch-1.0 -vf filesrc location=/home/focus/workspace_focus/full/hevc/ra_main-BasketballDrillText_832x480_50_qp22.bin  ! h265parse ! vaapih265dec ! "video/x-raw(memory:DMABuf)" ! vaapipostproc ! "video/x-raw(memory:VASurface)" ! vaapih264enc ! h264parse ! filesink location=test1.h264  (failed)
gst-launch-1.0 -vf filesrc location=/home/focus/workspace_focus/full/hevc/ra_main-BasketballDrillText_832x480_50_qp22.bin  ! h265parse ! vaapih265dec ! "video/x-raw(memory:DMABuf)" ! vaapipostproc ! "video/x-raw" ! vaapih264enc ! h264parse ! filesink location=test1.h264 (failed)

XinfengZhang, 2020-12-11:
is it a regression ?   especially before fe06066f8c643b75d6cdac21df8e16106d74e89c

Jister, 2020-12-24:
@FocusLuo Could you try intel-media-20.4 branch with latest gmmlib?

yefengxx, 2021-01-12:
I tried intel-media-20.4 branch with latest gmmlib, but it still failed. 
**Gmmlib:**
commit 6f15b795e1511febf233b42d7dde603f52fcfdab (HEAD -> master, tag: intel-gmmlib-20.4.1, origin/master, origin/HEAD)
Author: johnbash <johnbasha.shaik@intel.com>
Date:   Tue Dec 22 15:22:38 2020 +0530

Fix for Media corruption issue with the new interface (#15)
**Media-driver:**
commit 74e2f11131def34314ce52940e8d0028a4aba01c (HEAD -> intel-media-20.4, tag: intel-media-20.4.5, origin/intel-media-20.4)
Author: Sherry Lin <sherry.lin@intel.com>
Date:   Tue Dec 29 16:56:05 2020 +0800

    Bump up media version to 20.4.5

**cmd-line:**
root@media-ci-agent-tgl8:/home/yefeng/yuankun_test# gst-launch-1.0 -vf filesrc location=../media/hevc/ra_main-BasketballDrillText_832x480_50_qp22.bin ! h265parse ! vaapih265dec ! "video/x-raw(memory:DMABuf)" ! fakesink

**ERROR:** from element /GstPipeline:pipeline0/GstH265Parse:h265parse0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbaseparse.c(3675): gst_base_parse_loop (): /GstPipeline:pipeline0/GstH265Parse:h265parse0:
streaming stopped, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.

Jister, 2021-01-14:
@yefengxx I have tried to reproduce the issue locally, but cannot reproduce it, the case can pass both with the 20.4 branch and latest master branch, could you help to double confirm? My test command as follows:
gst-launch-1.0 -vf filesrc location=./ra_main-BasketballDrillText_832x480_50_qp22.bin ! h265parse ! vaapih265dec ! "video/x-raw(memory:DMABuf)" ! fakesink

FocusLuo, 2021-01-23:
@yefengxx @Jister let us to have a sync on this in next Monday or Tuesday?

Jister, 2021-01-25:
@FocusLuo sure

cqian2, 2022-03-07:
@Sherry-Lin , seems that the failure happens in  vaapih264enc, thus maybe codec side can help check it?

XinfengZhang, 2022-03-08:
there are two much fixes related with it , suppose it has been already fixed , and as I know it should already work. @FocusLuo  could you help to confirm?

FocusLuo, 2022-03-10:
thanks all, closed, if still have other issue let us to open new one.

closed 2022-03-10 by FocusLuo