intel/media-driver #1105: Intel i5 6300U missing vp9 profile

taktiker12, 2020-12-09:
thank you very much for this code , i just installed ubuntu 20.04 , kernel-5.4 running fine on thinkpad x260 i( 6300U intel HD 520 . 
output of vainfo shows this
libva info: VA-API version 1.10.0
libva info: User environment variable requested driver 'i965'
libva info: Trying to open /usr/local/lib/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_10
/usr/local/lib/dri/hybrid_drv_video.so init failed
Not using hybrid_drv_video.so
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.10 (libva 2.10.0.pre1)
vainfo: Driver version: Intel i965 driver for Intel(R) Skylake - 2.4.0.pre1 (2.3.0-31-gd87db21)
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointStats
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointStats
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointStats
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
-----------------------------------------------------------------------------------------------
i followed the istructions from https://gist.github.com/Brainiarc7/24de2edef08866c304080504877239a3

i hope to find a solution here 
regards
mounir

XinfengZhang, 2020-12-11:
what's the problem? you installed i965 driver ,not media-driver. so you want to install media-driver?

taktiker12, 2020-12-11:
thank Xinfeng
with media driver it does not show vp9 too .

taktiker12, 2020-12-18:
hi XinfengZhang
i have installed media-driver but vainfo does not show vp9 profile .
PS: i put iHD in /etc/environement

XinfengZhang, 2020-12-21:
it is not an real issue, SKL does not support VP9 , please refer to https://github.com/intel/media-driver#decodingencoding-features

closed 2020-12-31 by XinfengZhang