intel/media-driver #1108: AV1 decoder has corruption inside the stream.

HeJunyan, 2020-12-15:
The AV1 decoder can not successfully decode the stream:

data/00000527.ivf 
in
https://code.videolan.org/videolan/dav1d-test-data/-/tree/master/8-bit

both gstreamer pipeline:

gst-launch-1.0 -v -f filesrc location=~/dav1d-test-data/8-bit/data/00000527.ivf ! ivfparse ! vaapiav1dec ! checksumsink2 file-checksum=false frame-checksum=false plane-checksum=false dump-output=true dump-location=./tt.yuv

and MSDK cmd:
sample_decode av1 -i ~/dav1d-test-data/8-bit/data/00000527.ivf  -o out.yuv

generate the same result. The output begins to have corruption form the 11th frame.

dmitryermilov, 2020-12-15:
@wangyan-intel , FYI

wangyan-intel, 2020-12-15:
@Jexu Could you please take a look? Thanks.

rosetta-jpn, 2021-01-08:
Any update on this?

Jexu, 2021-01-08:
Hi @rosetta-jpn ,
it maybe driver issue and still under debug.

Jexu, 2021-01-14:
@rosetta-jpn 
can you try the latest master code and we have a fix 2ccc52ed0081f33d0b2a6f57e7d3b51eadb44270 in the master which can fix above corruption now.

rosetta-jpn, 2021-01-14:
Thanks.
Can you tell me the root cause and how likely there are videos causing the issue?
Is it quite edge case?

Jexu, 2021-01-15:
@rosetta-jpn 
It is caused by CTF buffer.  For av1, each frame has chance to generate one ctf buffer, these ctf buffer maybe transfered and used by rest frame. But in current driver, the ctf buffer will be removed along with reference frame even this ctf buffer is still needed by rest frame. So it will cause incorrect decoding.

rosetta-jpn, 2021-01-15:
Thanks. I think you mean CDF (Cumulative Distribution Function) by CTF.
The codec feature is probably so usual in av1 streams that the fix is essential in our product.

HeJunyan, 2021-02-18:
In fact, @rosetta-jpn and hiroh@chromium.org reported this issue, can you help to verify this? Thanks.

XinfengZhang, 2021-03-26:
I close the issue, because it already been fixed by https://github.com/intel/media-driver/commit/2ccc52ed0081f33d0b2a6f57e7d3b51eadb44270, any concern about it , you could re-open it

closed 2021-03-26 by XinfengZhang