intel/media-driver #1116: sample_decode may has a bug when running in multi-process.

wenbinc-Bin, 2020-12-25:
I run simple_2_decode_vmem of MediaSDK to decode 1080p video on the DG1 platform with 8GB video memory. when I run it in 120 processes it will be blocked at somewhere, but when I run it in 100 processes it works well and will be finished in 5 minutes. If I  force it to allocate 32 surfaces (set Request.NumFrameSuggested to 32 before it call alloc() ) and run simple_2_decode_vmem in 50 processes, some processes will be blocked and some will report GPU_HANG error when calling DecodeFrameAsync().

I can also produce this issue on sample_decode (command line: sample_decode h264 -device /dev/dri/renderD128 -vaapi -hw -i video_1920x1080.264 -o /dev/null ). After several testing I found that the error only occur when I run program in multi-process and the video memory used is over 4GB and only on the platform with 8GB video memory (DG1 with 4GB works fine).

wangyan-intel, 2020-12-25:
Please submit MSDK issue to MSDK repo: https://github.com/Intel-Media-SDK/MediaSDK/issues. Thanks.

wenbinc-Bin, 2020-12-25:
Ok, Thanks.

closed 2020-12-25 by wenbinc-Bin