intel/media-driver #1117: build media-driver-20.4 error

zhangyukai0302, 2020-12-29:
media_driver/media_driver_next/agnostic/common/os/mos_utilities_next.h:132:13:错误：检测到调用delete运算符时可能出现的问题：[-Werror] 
          delete(ptr)
media_driver/media_driver_next/agnostic/common/os/mos_utilities_next.h:132:13:错误：对不完全的类型‘class vp::VpPipeline’的非法使用[-Werror]
In file include from media_driver/include/common/media_interfaces/media_interfaces.cpp:33:0:
media_driver/agnostic/common/media_interfaces/media_interfaces_vphal.h:35:7:错误：‘class vp::VpPipeline’的向前声明[-Werror]

XinfengZhang, 2020-12-29:
@zhangyukai0302 , could you provide more details? about the env and compiler?

zhangyukai0302, 2020-12-29:
gcc 4.8.5 (Red hat 4.8.5-39)
centos 7.7.1908

XinfengZhang, 2020-12-29:
I suppose the gcc version is too older, gcc > 6.3 was recommended. another question: could you compile master branch on your machine?

zhangyukai0302, 2020-12-29:
master is same mistake.

zhangyukai0302, 2020-12-29:
刚安装了gcc8.3还是报一样的错误

Sherry-Lin, 2021-09-13:
Does this issue still exist?

Sherry-Lin, 2021-11-23:
no response from submitter couple months so close it.

closed 2021-11-23 by Sherry-Lin