intel/media-driver #1118: Kernel version for TGL

dmitryermilov, 2020-12-30:
Hi @XinfengZhang , @Sherry-Lin , 
What is minimal kernel version which fully supports TGL and recommended to production ?  What is the right process to report TGL GPU hang issues observed on Ubuntu 20.04 built-in kernel ? Report them to media-driver project? Or to report them to Ubuntu maintainers?  Or to report them to drm-tip maintainers? Or something else? 
There are bugs like https://github.com/Intel-Media-SDK/MediaSDK/issues/2554 plus a couple of other threads and I expect number of such issues will be increased.

XinfengZhang, 2020-12-31:
as I know, not all TGL KMD patch was upstreamed , so we did not use built-in kernel to verify TGL. I suppose we should prepare a 
 github wiki page to describe the situation and suitable kernel version.  user should choose recommended kernel for new platforms. it maybe is not real issue, just issue with the process and upstream timeline.

dvrogozh, 2020-12-31:
Concerning Ubuntu, you can try to use `apt-get install linux-oem` kernel running Ubuntu 20.04.1. One of the delta with default ubuntu kernel is new devices support. Read more here: https://wiki.ubuntu.com/Kernel/OEMKernel. In particular TGL support was added into this kernel for Ubuntu 20.04.1, see https://bugs.launchpad.net/intel/+bug/1855399. And refer to https://wiki.ubuntu.com/FocalFossa/ReleaseNotes/ChangeSummary/20.04.1 (search for TGL).

Sherry-Lin, 2021-03-02:
Based on execution data, 5.9+ kernel could support TGL very well.