intel/media-driver #1119: Mpeg2 decoder crash because of missing reference.

HeJunyan, 2020-12-30:
The gstreamer pipeline 
gst-launch-1.0 -vf filesrc location=~/koncercik.mpg ! parsebin ! vaapimpeg2dec ! fakesink
crashes because missing the reference.

This stream have group of picture header. It is an optional header that can be used immediately before a coded I-frame to indicate to the decoder if the first consecutive B-pictures immediately following the coded I-frame can be reconstructed properly in the case of a random access.
   
In that case, the B frames may miss the previous reference and can still be correctly decoded. 
We set   forward_reference_picture = VA_INVALID_ID, but driver crashes when access the reference frame.

Even we really misses the reference frame, some manner such as inserting grey picture should be used to handle these cases, rather than crash.

HeJunyan, 2020-12-31:
CC: @XinfengZhang & @xhaihao

XinfengZhang, 2020-12-31:
two questions:
1.  you are talking about IBBBB, why forward reference picture is invalid?  backward reference picture should be invalid
2.  the consecutive B frame just have one reference frame?

HeJunyan, 2020-12-31:
Yes, the B is in the stream header, and has no forward reference frame. I can send you the stream to analyze it if you want.
And, more important, even the B frame missing the forward_reference_picture  or back_reference_picture, and we set VA_INVALID_ID for these field, the driver should not crash.

XinfengZhang, 2020-12-31:
> Yes, the B is in the stream header, and has no forward reference frame. I can send you the stream to analyze it if you want.
> And, more important, even the B frame missing the forward_reference_picture or back_reference_picture, and we set VA_INVALID_ID for these field, the driver should not crash.

you mean, the driver should not crash, but could add some protection code (condition check) and return failure?

HeJunyan, 2020-12-31:
What is the behaviour of H264/265 and other codecs when missing some reference? The same behaviour is OK.

HeJunyan, 2020-12-31:
It's better to insert some. for example, grey surface for missing reference surface. That kind of B frame I mentioned can be decoded successfully with just back_reference_picture. So, return failure is not a good option.

XinfengZhang, 2020-12-31:
it means all of prediction is from back_reference_picture, no any prediction is from forward_reference_picture.  is it a common usage case?

Jexu, 2021-03-29:
> Yes, the B is in the stream header, and has no forward reference frame. I can send you the stream to analyze it if you want.
> And, more important, even the B frame missing the forward_reference_picture or back_reference_picture, and we set VA_INVALID_ID for these field, the driver should not crash.

Hi @HeJunyan, by checking the driver, when there is no forward or backward reference, driver will set the reference index to be destination frame. That means driver can always get non-null reference. can you share the bitstream to me and i can double check that.

PR #1278 (2021-11-01): [Decode] avoid the array out of bounds when the forward or backward reference frame is invalid
reference frame is invalid

fixes #1119
commit c782c8b0 (2021-11-01): [Decode] avoid the array out of bounds when the forward or backward
reference frame is invalid

fixes #1119
PR #1285 (2021-11-08): [Decode] avoid the array out of bounds when the forward or backward
reference frame is invalid

fixes #1119
commit f8d02055 (2021-11-08): [Decode] avoid the array out of bounds when the forward or backward
reference frame is invalid

fixes #1119
closed 2021-11-01 by guowillyy