intel/media-driver #1123: vaCopy implementation does not work on Gen9/11 platforms

aobolensk, 2021-01-11:
When vaCopy is called on Gen 9/11 platforms `McpyDevice::CreateFactory` function is called: https://github.com/intel/media-driver/blob/80dd4ffdb6cd4ff8885ab17983bc5dec060ce6c3/media_driver/linux/common/ddi/media_libva.cpp#L5990
This function fails because there is no McpyDevice created for Gen 9/11 platforms: https://github.com/intel/media-driver/blob/80dd4ffdb6cd4ff8885ab17983bc5dec060ce6c3/media_driver/linux/common/media_interfaces/media_interfaces.cpp#L304
```
$ grep -rn 'RegisterHal<McpyDevice'
media_driver/media_interface/media_interfaces_m12_tgllp/media_interfaces_g12_tgllp.cpp:388:    RegisterHal<McpyDeviceG12Tgllp>((uint32_t)IGFX_TIGERLAKE_LP);
media_driver/media_interface/media_interfaces_m12_dg1/media_interfaces_g12_dg1.cpp:56:    RegisterHal<McpyDeviceG12Tgllp>((uint32_t)IGFX_DG1);
```
This function should be either available for all platforms or we need to have query interface for it. Otherwise this function is not usable properly.

dmitryermilov, 2021-01-11:
@rawide , @XinfengZhang , @dvrogozh , we expected that vaCopy is a general interface working on **all** platforms. Otherwise, it's impossible to use it in solutions.

XinfengZhang, 2021-01-12:
how about to add a capability of vaCopy if there are no  "all platform" support?

dmitryermilov, 2021-01-12:
> how about to add a capability of vaCopy if there are no "all platform" support?

Hi Carl,
Yes, it's an option. It can help. But from usability perspective it would be great to avoid it. Originally AFAIR vaCopy was declared as unified copy solution.  Now it doesn't look this way. Usually most of media applications use different kind of frame copy functionality. If vaCopy doesn't work everywhere, applications needs to keep copy fallback code. So app will have to support two copy paths. It introduces additional complexity.