intel/media-driver #1124: JPEG decode to RGB fails on some images

artlav, 2021-01-13:
Some jpg files decode into line-like noise when the output surface is set to VA_RT_FORMAT_RGB32. (Most work fine).

Tested on:
- i5-6260U (Iris Graphics 540)
- m5-6Y54 (HD Graphics 515)
- i3-10110U (UHD Graphics)

Version:
- VA version: 1.10, vendor: Intel iHD driver for Intel(R) Gen Graphics - 21.1.0  (today's git compiled with debug)
- VA version: 1.10, vendor: Intel iHD driver for Intel(R) Gen Graphics - 20.4.5  (from distro's repository)

JPEG Baseline VLD config attributes do have RGB32 in their RT Format list.

Reproduction:  
Take tinyjpeg (libva-utils), set the output surface to RGB, run it on this file: https://orbides.org/img/sml1.jpg

```
//Add these to tinyjpeg_decode in tinyjpeg.c
surface_type = VA_RT_FORMAT_RGB32;
forcc.value.value.i = VA_FOURCC_ARGB;

//Just before this line
va_status = vaCreateSurfaces(va_dpy,surface_type,
   priv->width[scan_num],priv->height[scan_num], //alignment?
   &surface_id, 1, &forcc, 1);
```

Info on the test file.
I tried varying every parameter, and none of them are critical to causing the bug.
Some other files were 4:4:0, some optimized and some standard huffman tables, larger or smaller. 
So far i found 3-4 files that fail.
They open just fine in other tools, and load/saving them with Gimp with same parameters does not fix the problem.
```
ExifTool Version Number         : 12.00
File Name                       : sml1.jpg
Directory                       : .
File Size                       : 4.5 kB
File Modification Date/Time     : 2021:01:13 19:22:51+03:00
File Access Date/Time           : 2021:01:13 19:23:17+03:00
File Inode Change Date/Time     : 2021:01:13 19:23:11+03:00
File Permissions                : rw-r--r--
File Type                       : JPEG
File Type Extension             : jpg
MIME Type                       : image/jpeg
JFIF Version                    : 1.01
Resolution Unit                 : None
X Resolution                    : 1
Y Resolution                    : 1
Image Width                     : 160
Image Height                    : 120
Encoding Process                : Baseline DCT, Huffman coding
Bits Per Sample                 : 8
Color Components                : 3
Y Cb Cr Sub Sampling            : YCbCr4:4:4 (1 1)
Image Size                      : 160x120
Megapixels                      : 0.019
```

Another test file: https://orbides.org/img/bad_big.jpg
```
ExifTool Version Number         : 12.00
File Name                       : bad_big.jpg
Directory                       : .
File Size                       : 2.0 MB
File Modification Date/Time     : 2021:01:13 19:55:14+03:00
File Access Date/Time           : 2021:01:13 19:55:19+03:00
File Inode Change Date/Time     : 2021:01:13 19:55:14+03:00
File Permissions                : rw-r--r--
File Type                       : JPEG
File Type Extension             : jpg
MIME Type                       : image/jpeg
JFIF Version                    : 1.01
Resolution Unit                 : inches
X Resolution                    : 72
Y Resolution                    : 72
Image Width                     : 2736
Image Height                    : 3648
Encoding Process                : Baseline DCT, Huffman coding
Bits Per Sample                 : 8
Color Components                : 3
Y Cb Cr Sub Sampling            : YCbCr4:4:0 (1 2)
Image Size                      : 2736x3648
Megapixels                      : 10.0
```

More details are visible on a larger file.
Source image: https://orbides.org/img/refx.jpg (also a test file)
Output (resized 33%): https://orbides.org/img/pretile.jpg
Output after i tried to detile it: https://orbides.org/img/tiles.jpg

Basically it outputs tiled buffer as is with pieces missing (maybe, not sure if i applied the right detiling), and in some YUV format but without the Cr and Cb parts (if you vaDeriveImage+vaMapBuffer it and try reading past the reported buffer size there is something that looks like Cr and Cb parts).
Feels like something does not set something properly somewhere, or fails to check for capabilities correctly.

Same effect can be achieved by trying to decode any image larger than 4096 pixels in width or height (config attributes report max image size to be 16384x16384 on all systems i tried).
But other than that, i wasn't able to find any pattern to which files fail and which don't.

Another thing i noticed is that running with debug prints, some lines are missing on broken images.
No idea what this means.

Bad image snippet:
```
[LIBVA]:VERBOSE  - DdiMediaUtil_AllocateBuffer:825: Alloc  2114056 bytes resource.
[CODEC]:ENTER    - Execute
[CODEC]:ENTER    - Execute
[CODEC]:ENTER    - SetFrameStates
[CODEC]:ENTER    - InitializeBeginFrame
[CODEC]:ENTER    - CheckAndInitialize
[CODEC]:ENTER    - DumpPicParams
[CODEC]:ENTER    - DumpScanParams
[CODEC]:ENTER    - DumpHuffmanTable
[CODEC]:ENTER    - DumpIQParams
[CODEC]:ENTER    - VerifySpaceAvailable
[CODEC]:ENTER    - DecodeStateLevel
[CODEC]:ENTER    - SendPrologWithFrameTracking
[CODEC]:ENTER    - IsMmcEnabled
[CODEC]:ENTER    - SetPipeBufAddr
[CODEC]:ENTER    - StartStatusReport
[CODEC]:ENTER    - AddSfcCommands
[CODEC]:ENTER    - DecodePrimitiveLevel
[CODEC]:ENTER    - EndStatusReport
[CODEC]:ENTER    - UpdateUserFeatureKey
[CODEC]:ENTER    - ResetStatusReport
[CODEC]:ENTER    - EndFrame
[CODEC]:VERBOSE  - EndFrame:891: NumReportsAvailable = 1
[LIBVA]:EXIT     - EndPicture: eStatus = 0x0
[LIBVA]:EXIT     - DdiDecode_EndPicture: eStatus = 0x0
[LIBVA]:ENTER    - DdiMedia_SyncSurface
[LIBVA]:ENTER    - DdiMedia_StatusCheck
[CODEC]:ENTER    - GetStatusReport
```

Good image snippet:
```
[LIBVA]:VERBOSE  - DdiMediaUtil_AllocateBuffer:825: Alloc   642702 bytes resource.
[CODEC]:ENTER    - Execute
[CODEC]:ENTER    - Execute
[CODEC]:ENTER    - SetFrameStates
[CODEC]:ENTER    - InitializeBeginFrame
[CODEC]:ENTER    - CheckAndInitialize
[CODEC]:ENTER    - IsSfcOutputSupported
[CODEC]:ENTER    - Initialize
[CODEC]:ENTER    - AllocateResources
[CODEC]:ENTER    - DumpPicParams
[CODEC]:ENTER    - DumpScanParams
[CODEC]:ENTER    - DumpHuffmanTable
[CODEC]:ENTER    - DumpIQParams
[CODEC]:ENTER    - VerifySpaceAvailable
[CODEC]:ENTER    - DecodeStateLevel
[CODEC]:ENTER    - SendPrologWithFrameTracking
[CODEC]:ENTER    - IsMmcEnabled
[CODEC]:ENTER    - SetPipeBufAddr
[CODEC]:ENTER    - StartStatusReport
[CODEC]:ENTER    - AddSfcCommands
[CODEC]:ENTER    - SetSfcStateParams
[CODEC]:ENTER    - UpdateInputInfo
[CODEC]:ENTER    - SetSfcIefStateParams
[CODEC]:ENTER    - DecodePrimitiveLevel
[CODEC]:ENTER    - EndStatusReport
[CODEC]:ENTER    - UpdateUserFeatureKey
[CODEC]:ENTER    - ResetStatusReport
[CODEC]:ENTER    - EndFrame
[CODEC]:VERBOSE  - EndFrame:891: NumReportsAvailable = 1
[LIBVA]:EXIT     - EndPicture: eStatus = 0x0
[LIBVA]:EXIT     - DdiDecode_EndPicture: eStatus = 0x0
[LIBVA]:ENTER    - DdiMedia_SyncSurface
[LIBVA]:ENTER    - DdiMedia_StatusCheck
[CODEC]:ENTER    - GetStatusReport
```

XinfengZhang, 2021-01-18:
seems the only difference is there are log related with SFC call, it means there are a color space conversion or scaling after decoding. no such operation with the failed case. but it was strange, just like your description, only part of the jpeg picture failed, so I suspect it was because the format of jpeg file itself. for example: if the jpeg is 420, but output is set to ARGB. then it will call SFC internally. if jpeg already is RGB , no additional SFC needed.  thanks for reporting, we will check it.

Jexu, 2021-03-29:
@artlav 
just checked your clip https://orbides.org/img/sml1.jpg
the chroma type for this jpeg file is yuv 444 which means you cannot just set the surface format to be RGB directly. media driver always creates the render surface by the bitstream chrome type. if you want to decode it into RGB format, you may need to use decode SFC for color converting

artlav, 2021-03-29:
@Jexu
Problem is, most files work just fine, whether they are yuv 444, yuv 440 or anything. If this is unsupported, then i expect it to never work or give an error, if it's automatically converted then i expect it to always work. It's the inconsistency that is the problem.

Jexu, 2021-03-30:
will add the surface format check for invalid setting.

Jexu, 2024-02-01:
The issue has been fixed in commit f3e38626762ddb1db62b8ab66d392f885342868a by adding strict format check.

Close this issue now and please re-open it again if hvaing any other question or concern.

closed 2024-02-01 by Jexu