intel/media-driver #1125: Failed to schedule to VDBOX2 in some decoding cases

joojler, 2021-01-19:
When we created multiple decoding sessions on platforms with 2 VDBOXs, after a while, the decoding workload can't be scheduled to VDBOX2. After some investigation, I found it's a bug in media-driver. The reasons are as below:

1. media-driver create & destroy the node association in function Mos_Specific_CreateVideoNodeAssociation & Mos_Specific_DestroyVideoNodeAssociation which are located in ./media_driver/linux/common/os/mos_os_specific.c
2. Due to incorrect macro definition in media_driver/linux/common/os/mos_os_specific.h for MOS_GPU_NODE_MAX, both MOS_GPU_NODE_MAX and MOS_GPU_NODE_VIDEO2 equals 7. 
typedef enum _MOS_GPU_NODE
{
    MOS_GPU_NODE_3D      = I915_EXEC_RENDER,
    MOS_GPU_NODE_COMPUTE = (5<<0), //To change to compute CS later when linux define the name
    MOS_GPU_NODE_VE      = I915_EXEC_VEBOX,
    MOS_GPU_NODE_VIDEO   = I915_EXEC_BSD,
    MOS_GPU_NODE_VIDEO2  = I915_EXEC_VCS2,
    MOS_GPU_NODE_BLT     = I915_EXEC_BLT,
    **MOS_GPU_NODE_MAX     = 7**,//GFX_MAX(I915_EXEC_RENDER, I915_EXEC_VEBOX, I915_EXEC_BSD, I915_EXEC_VCS2, I915_EXEC_BLT) + 1
} MOS_GPU_NODE, *PMOS_GPU_NODE;

#define I915_EXEC_VCS2                   (7<<0)
3. Above mistake will cause the CodechalDecode::~CodechalDecode() failure for VDBOX2 according to the following logic
[image]
4. Then the association on VDBOX2 will never be destroyed. The result is the workload can't be able to scheduled to VDBOX2
5. By modifying the MOS_GPU_NODE_MAX to GFX_MAX(I915_EXEC_RENDER, I915_EXEC_VEBOX, I915_EXEC_BSD, I915_EXEC_VCS2, I915_EXEC_BLT) + 1, the issue is fixed. 

Please help to suggest if this fix is valid? thanks.

Jexu, 2021-01-25:
Hi @joojler, You can try to change value of I915_EXEC_VCS2 to 6<<0 since this value is idle

joojler, 2021-01-27:
Got it. Just wondering the code for proper MOS_GPU_NODE_MAX definition as GFX_MAX(I915_EXEC_RENDER, I915_EXEC_VEBOX, I915_EXEC_BSD, I915_EXEC_VCS2, I915_EXEC_BLT) + 1 is actually there previously. Any possible bugs reported before for this kind of definition?

Jexu, 2021-02-02:
We only have 6 gpu node definition here, so MOS_GPU_NODE_MAX should be equal to 7. The value of the gpu node here seems not serial, i guess GFX_MAX+1 will not show the correct value when I915_EXEC_VCS2=7<<0

joojler, 2021-03-18:
Hi Jexu, submitted an new PR with the modification of I915_EXEC_VCS2=6<<0.

Jexu, 2021-04-07:
@joojler 
can you close this #1126 since you have sent new one #1164
and we will help merge the later one

Jexu, 2021-05-07:
@XinfengZhang can you close this issue since the PR is merged.

closed 2021-05-07 by XinfengZhang