intel/media-driver #1127: Failed to create a RGBX VASurface importing a GBM buffer

harryshan, 2021-01-20:
Hardware platform: Intel HD630
media-driver version: Compiled from Jan.14 commit (13c0e1480a96510ca1d3fff9f2e8574e1fba1979)

VA trace for configurations: 
[vatest.064422.thd-0x0000714e.txt](https://github.com/intel/media-driver/files/5840569/vatest.064422.thd-0x0000714e.txt)

What I'm trying to do is creating a RGBX buffer using minigbm, importing the buffer to a VASurface and also an x11 Pixmap so that they're shared. 
Then I'll decode H264 to the RGBX VASurface following SFC sample, using ```VAProcPipelineParameterBuffer.additional_outputs```, to avoid a ```vaPutSurface``` call, in order to eleminate memory copy and save power.

Questions:
1. Do my demands make sense in terms of power saving? Does SFC conversion use less power than vaPutSurface?
2. Is there anything wrong in my configuration?

Any help is much appreciated

XinfengZhang, 2021-01-29:
1. yes 2. we already discussed in the mail loop, there are progress, after all done, I will paste the key changes into this issue to help other users.

XinfengZhang, 2021-03-09:
close the issue, because it already be fixed though some email communications

closed 2021-03-09 by XinfengZhang