intel/media-driver #1129: gmm errors while building the media-driver on Yocto

vijnasu, 2021-01-21:
Is there a fix for the following error. I am trying to build the driver for two days.

```
media_libva_util.cpp:175:5: error: 'GMM_RESCREATE_CUSTOM_PARAMS' was not declared in this scope; did you mean 'GMM_RESCREATE_PARAMS'?

media_libva_util.cpp:503:29: error: 'gmmCustomParams' was not declared in this scope

error: 'GMM_CLIENT_CONTEXT' {aka 'class GmmLib::GmmClientContext'} has no member named 'CreateCustomResInfoObject'; did you mean 'CreateResInfoObject'?

 error: 'using MEDIA_SYSTEM_INFO = using MEDIA_GT_SYSTEM_INFO = GT_SYSTEM_INFO' {aka 'struct GT_SYSTEM_INFO'} has no member named 'NumThreadsPerEu'
  156 |         gtSystemInfo->ThreadCount       = gtSystemInfo->EUCount * gtSystemInfo->NumThreadsPerEu;

```

XinfengZhang, 2021-01-22:
@vijnasu  have you update gmmlib https://github.com/intel/gmmlib? this new definition is from 1598c1b46fb911cb9060d1a5380aec1ca1ef2c19

XinfengZhang, 2021-01-22:
I add this issue to "enhancement in plan", we should have a table for media-driver compliant gmmlib , better in wiki page.

troy818, 2021-03-08:
@vijnasu I have the same problem with you and I've fixed it. After building gmmlib, I do "sudo make install" in the gmmlib build folder. Then the error is solved.

@XinfengZhang Hi, I think maybe the "Install" session of Gmmlib README add "$ sudo make install"?  I think maybe it will be clearer for the user like me. Thanks.

dvrogozh, 2021-03-08:
https://github.com/intel/media-driver#building says:
```
2. Build and install gmmlib master
```

That's up to gmmlib to describe their build and install procedure. Media driver readme focus on the driver part of the story.

closed 2021-03-09 by XinfengZhang